gtri/scrimmage #3: make docs sphinx_git dependency

laurathepluralized, 2017-08-08:
`make docs` fails if sphinx_git is not installed: 
```
$ make docs
Scanning dependencies of target sphinx-docs
[ 50%] Building HTML documentation with Sphinx

Extension error:
Could not import extension sphinx_git (exception: No module named sphinx_git)
docs/CMakeFiles/sphinx-docs.dir/build.make:57: recipe for target 'docs/CMakeFiles/sphinx-docs' failed
make[3]: *** [docs/CMakeFiles/sphinx-docs] Error 1
CMakeFiles/Makefile2:3732: recipe for target 'docs/CMakeFiles/sphinx-docs.dir/all' failed
make[2]: *** [docs/CMakeFiles/sphinx-docs.dir/all] Error 2
CMakeFiles/Makefile2:3675: recipe for target 'docs/CMakeFiles/docs.dir/rule' failed
make[1]: *** [docs/CMakeFiles/docs.dir/rule] Error 2
Makefile:951: recipe for target 'docs' failed
make: *** [docs] Error 2
```

It looks like running `setup/install-binaries.sh` before starting the rest of the scrimmage build process didn't  install sphinx_git, but after I ran `sudo python2.7 -m pip install sphinx_git`, `make docs` worked fine.

SyllogismRXS, 2017-08-08:
At the end of install-binaries.sh is the line: pip install sphinx-git. Does that not install sphinx-git for you?
(Note: We need to improve our install-binaries.sh script to handle more configurations.)

laurathepluralized, 2017-08-10:
I thought I had run install-binaries before trying to recompile docs when I got this error, but it's possible it slipped my mind. As a test, I uninstalled sphinx-git via pip and then ran install-binaries.sh again, and I saw pip reinstall sphinx-git. And while sphinx-git was uninstalled, trying to make docs gave me no errors about sphinx_git not being installed. 

I'm not sure what was causing that error earlier, then. I'll see if I can replicate (and better document) this issue later this week when I do a clean install of scrimmage on another machine, and I'll update this issue then. Thanks!

laurathepluralized, 2017-08-12:
I tried again on my other machine, and was careful to make sure that I ran install-binaries.sh before running `make docs`. I'm getting the same error again, even though pip definitely ran at the end of install-binaries. Here's the output from install-binaries.sh:
```
$ sudo ./setup/install-binaries.sh
[sudo] password for <my-username>: 
---------------------------------------------------
Detecting Linux operating system variant.
This is Ubuntu. Using dpkg.
Detecting which required packages are not installed.
All dependencies are installed. No further action is required.
The directory '/home/<my-username>/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/<my-username>/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already satisfied (use --upgrade to upgrade): sphinx-git in /usr/local/lib/python3.5/dist-packages
Requirement already satisfied (use --upgrade to upgrade): six in /usr/lib/python3/dist-packages (from sphinx-git)
Requirement already satisfied (use --upgrade to upgrade): sphinx in /usr/lib/python3/dist-packages (from sphinx-git)
Requirement already satisfied (use --upgrade to upgrade): GitPython>=0.3.6 in /usr/local/lib/python3.5/dist-packages (from sphinx-git)
Requirement already satisfied (use --upgrade to upgrade): Jinja2>=2.3 in /usr/lib/python3/dist-packages (from sphinx->sphinx-git)
Requirement already satisfied (use --upgrade to upgrade): Pygments>=2.0 in /usr/lib/python3/dist-packages (from sphinx->sphinx-git)
Requirement already satisfied (use --upgrade to upgrade): docutils>=0.11 in /usr/lib/python3/dist-packages (from sphinx->sphinx-git)
Requirement already satisfied (use --upgrade to upgrade): babel!=2.0,>=1.3 in /usr/lib/python3/dist-packages (from sphinx->sphinx-git)
Requirement already satisfied (use --upgrade to upgrade): alabaster<0.8,>=0.7 in /usr/lib/python3/dist-packages (from sphinx->sphinx-git)
Requirement already satisfied (use --upgrade to upgrade): sphinx_rtd_theme<2.0,>=0.1 in /usr/lib/python3/dist-packages (from sphinx->sphinx-git)
Requirement already satisfied (use --upgrade to upgrade): gitdb2>=2.0.0 in /usr/local/lib/python3.5/dist-packages (from GitPython>=0.3.6->sphinx-git)
Requirement already satisfied (use --upgrade to upgrade): MarkupSafe in /usr/lib/python3/dist-packages (from Jinja2>=2.3->sphinx->sphinx-git)
Requirement already satisfied (use --upgrade to upgrade): pytz>=0a in /usr/lib/python3/dist-packages (from babel!=2.0,>=1.3->sphinx->sphinx-git)
Requirement already satisfied (use --upgrade to upgrade): smmap2>=2.0.0 in /usr/local/lib/python3.5/dist-packages (from gitdb2>=2.0.0->GitPython>=0.3.6->sphinx-git)
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```
Am I right in that this indicates that pip is only checking for sphinx-git and its dependencies for python3? I tried running `import sphinx_git` in both python2.7 and python3, and only python3 found it. Does BUILD_DOCS depend on the python2.7 sphinx_git? 

If that's the case, I'm not sure whether the best way to proceed would be to tell pip to install the python2.7 version of sphinx-git (probably the easiest option--just a one-line change in install-binaries.sh) or whether there is a way make docs can be made aware of the python3 option. If you want me to try fixing this and doing a pull request, or if you want me to test anything out, let me know.

Thanks!

SyllogismRXS, 2018-05-25:
I added a CMake function that checks for Python modules. If the sphinx_git python module doesn't exist, cmake will warn you now.

closed 2018-05-25 by SyllogismRXS