gtri/scrimmage #9: Python version specification

laurathepluralized, 2017-08-30:
The README says that, if I want to specify a particular Python version, I can set Python_ADDITIONAL_VERSIONS when I run cmake. However, this variable doesn't seem to be used in CMakeLists.txt, or anywhere else. Is there some other way to require the use of a specific Python version, and/or is this a planned enhancement? I went into the CMakeCache.txt and modified paths, but would prefer a simpler method if it is feasible. Thanks!

SyllogismRXS, 2017-08-30:
It looks like that functionality was old. I updated the cmake project to allow a developer to specify a PYTHON_MIN_VERSION variable. I tested it with Python 2.7 and 3.4 and it seemed to work.  If you don't specify the version, it defaults to 2.7. I updated the Python section of the README.md, so check that out. Try out the latest master commit to see if this fixes your problem.

laurathepluralized, 2017-08-30:
I just tried it out, and when I set PYTHON_MIN_VERSION=3.5, CMake ran successfully and found Python 3.5. Thanks so much!

SyllogismRXS, 2017-08-31:
I updated the cmake project template to use scrimmage core's python version.

closed 2017-08-31 by SyllogismRXS