gtri/scrimmage #14: generate_scenarios cannot find SCRIMMAGE_ROOT

laurathepluralized, 2017-09-14:
This issue is regarding scripts/generate_scenarios.py.

In the qsub function, generate_scenarios.py looks for SCRIMMAGE_ROOT, which is not defined in the current version of scrimmage. Here's the line:

```
scrimmage_root = os.getenv('SCRIMMAGE_ROOT')
```

My current workaround is defining SCRIMMAGE_ROOT before the script is run. For instance, in a shell script, I have a line like the following (with python3 for my most recent pull request, python2.7 otherwise): 

```
(export SCRIMMAGE_ROOT=/path/to/scrimmage && source ~/.scrimmage/setup.bash && python3 ${SCRIMMAGE_ROOT}/scripts/generate_scenarios.py --ranges therangesfile.xml --num_runs=100 --nodes '1' /path/to/mission/file.xml /path/to/results/directory)
```

I haven't found any environment variables in my scrimmage setup that provide the path to scrimmage the way SCRIMMAGE_ROOT is supposed to. Is there a more-correct way to tell generate_scenarios.py where to find the scrimmage repo than my workaround? Thanks!

laurathepluralized, 2017-09-15:
Somewhat-related note: SCRIMMAGE_ROOT is also required in scripts/scrimmage.sh, though apparently instead of the path to the scrimmage repo, the path scrimmage.sh is expecting from SCRIMMAGE_ROOT would be scrimmage's build directory (it is looking for /bin/scrimmage).

esquires, 2018-05-31:
I think this was fixed a while ago but we missed this ticket. Closing for now but if it is still an issue feel free to reopen.

closed 2018-05-31 by esquires