gtri/scrimmage #37: mistakes in some codes of tutorials

luanyun, 2018-03-29:
As a potential third-party developer, I found it is difficult to realize my designation follow tutorials and  API document.
For some mistakes in tutorials and various parameters which i don't know how to use.

eg. the last code segment in _Create a Sensor Plugin._
maybe it should be write as below? I can't make sure.

    for (auto kv : parent_->sensors()) {
        if (kv.first == "MyNoisyState0") {
            // Take a measurement from the MyNoisyState sensor
            auto msg = kv.second->sense<sc::State>(t);
            if (msg) {
                own_state = msg->data;
            }
        }
    }

luanyun, 2018-03-29:
And how to debug a plugin using Qt or other platform.
Are there any other suggestions for me ?

esquires, 2018-03-30:
Thanks for the suggestion. I have updated the documentation to reflect the fix in 87315df on master

closed 2018-03-30 by esquires