gtri/scrimmage #39: OpenAI confuse

luanyun, 2018-03-30:
Today When trying to Install SCRIMMAGE Python Bindings and Create an OpenAI Plugin follow the tutorial.

when running the /path/to/scrimmage/python/test/test_openai.py, An error come into forth "AttributeError: 'ScrimmageEnv' object has no attribute '_render'"

when running the created test_openai.py referred by the tutorial, I cannot register the environment
Traceback (most recent call last):
  File "/home/lht/scrimmage/my-scrimmage-plugins/test/test_openai.py", line 39, in <module>
    test_openai()
  File "/home/lht/scrimmage/my-scrimmage-plugins/test/test_openai.py", line 21, in test_openai
    env = gym.make('scrimmage1-v0')
  File "/home/lht/anaconda3/lib/python3.6/site-packages/gym/envs/registration.py", line 167, in make
    return registry.make(id)
  File "/home/lht/anaconda3/lib/python3.6/site-packages/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/home/lht/anaconda3/lib/python3.6/site-packages/gym/envs/registration.py", line 86, in make
    env = cls(**self._kwargs)
  File "/home/lht/scrimmage/scrimmage/python/scrimmage/external_control.py", line 98, in __init__
    self.scrimmage_process = self._start_scrimmage(False, True)
  File "/home/lht/scrimmage/scrimmage/python/scrimmage/external_control.py", line 203, in _start_scrimmage
    tree = ET.parse(self.mission_file)
  File "/home/lht/anaconda3/lib/python3.6/xml/etree/ElementTree.py", line 1196, in parse
    tree.parse(source, parser)
  File "/home/lht/anaconda3/lib/python3.6/xml/etree/ElementTree.py", line 597, in parse
    self._root = parser._parse_whole(source)
xml.etree.ElementTree.ParseError: junk after document element: line 9, column 0

luanyun, 2018-04-01:
have been solved.
for the first,  /path/to/scrimmage/python/scrimmage/external_control.py should be fixed by 'render'->'_render' in Ln 183
the second is for my fault. I forgot to register the plugin in src file.

esquires, 2018-04-26:
Thanks. This has been fixed in commit 8301bf9

closed 2018-04-26 by esquires