gtri/scrimmage #42: RayTrace does not seem to do anything

sciencectn, 2018-04-10:
It appears to return nothing: 
https://github.com/gtri/scrimmage/blob/master/src/plugins/sensor/RayTrace/RayTrace.cpp#L87

Also, the [cars-2d-maze mission](https://github.com/gtri/scrimmage/blob/master/missions/cars-2d-maze.xml) depends on the [AvoidWalls](https://github.com/gtri/scrimmage/blob/master/src/plugins/autonomy/AvoidWalls/AvoidWalls.cpp) plugin, and AvoidWalls doesn't work because it depends on RayTrace. If you run cars-2d-mission the cars just ignore the walls.

SyllogismRXS, 2018-05-22:
I think the Ray tracing is working, but the example AvoidWalls plugin is reading from the sensor plugin incorrectly. I'll take a look at this soon.

SyllogismRXS, 2018-05-25:
I updated the AvoidWalls plugin to subscribe to the correct RayTrace topic, so now the AvoidWalls plugin works as expected. I fixed this in commit e2f8c047d85c3ec810272bf34378e33f42b1ceea. Let me know if it works as expected now.

closed 2018-05-25 by SyllogismRXS