gtri/scrimmage #81: ExternalControlClient received bad grcp status

lvalcourt, 2018-06-12:
Error occurs when using more than two actors with OpenAI Autonomous plugin.

esquires, 2018-06-13:
This should be fixed in master.

If you are on v0.1.0-rc I would recommend using only 1 learner per scenario. You can used shared memory to pass around actions/observations to make it seem to python that it is controlling multiple agents.

On master there is better support for what you are looking to do and documentation will show up in about a day. If you need documentation before then you can do

```
cmake .. -DBUILD_DOCS=ON
make docs
firefox build/docs/sphinx/html/index.html
```

Also, see test/test_openai.py for examples of using discrete/continuous spaces as well as multi-agent scenarios with that are independent and combined.

Note that there is an api change on master from what you have been using and grpc is not used anymore.

closed 2018-06-13 by esquires

lvalcourt, 2018-06-13:
Sounds good. I am running master as of Sunday of this week. Will download documentation. 

Thanks