gtri/scrimmage #96: Multi-Agent Independant Interactions

lvalcourt, 2018-07-02:
By setting the number of agents and xml file in the python file, we can get the independant observations for each agent, given an independant action for each agent. However, a reward is return but in the form of a accumulative reward, meaning that the reward reflects the amount of reward of all agents together. I tried different settings (combine actors, global sensors) but it will always return a cumulative reward.  
Is there a way that the rewards can be recived as a array like the observation values?

esquires, 2018-07-02:
The openai api requires that a scalar is returned for reward.  The last return value of the environment's ``step`` function returns a dictionary called ``info`` that has an element ``return`` that has the individual returns.

lvalcourt, 2018-07-02:
Verified, thanks for the quick response and information.

closed 2018-07-02 by lvalcourt