gtri/scrimmage #97: Multi-Agent State/Observation

lvalcourt, 2018-07-03:
Given a multi-agent scenario, an array of x number of states is received. To be 100% sure, does these values, the states, represent a location in the environment for each agent or a x,y location specifically? I am tyring to eliminated position overlapping between agents. 

Thanks!

esquires, 2018-07-03:
If you are looking at ``contacts_`` then the state for each vehicle is defined in ``State.h`` which has x, y, z position and velocity as well as a quaternion that can be converted to euler angles.

lvalcourt, 2018-07-03:
I get the x,y,z location of the agent and convert it to lat/lon good inside the RLSimple class, to handle rewards and manage the movement of the agent. That is all good since I can calculate distances and overlapping between locations for locations visited by one agent. However, the autonomy class is created per agent, so now I am trying to detect position overlapp  between all of my agents. Is this possible? I am using the states of each agent on the script file to discover this but that is a bad assumption I beleive. 

An example of a State value isas follow for a two agent env: Obs: [0,18]

lvalcourt, 2018-07-03:
To be sure, the value that I am asking to know the definition of is the state value returned by each env.step(a).

Thanks!

closed 2018-07-05 by lvalcourt