gtri/scrimmage #100: ModuleNotFoundError: No module named 'scrimmage'

savasturk, 2018-07-04:
I followed this tutorial (http://www.scrimmagesim.org/docs/sphinx/html/tutorials/multiple-local-runs.html  )
everything is okey when apply this ":~/scrimmage/scrimmage/scripts$ ./scrimmage_runner.sh -t 100 -m ../missions/straight-vs-motorschemas.xml -p 7 " command. Then I received error like this: 
Traceback (most recent call last):
  File "generate_scenarios.py", line 45, in <module>
    import scrimmage.utils as utils
ModuleNotFoundError: No module named 'scrimmage'
Failed to generate scenarios

USAGE: 
          ./scrimmage_runner.sh [OPTIONS]

OPTIONS:
          -h   Display this usage information
          -t   Sets the number of simulation runs (tasks)
          -m   Mission file
          -r   Ranges file for generating new scenarios
          -p   Maximum number of parallel runs (tasks)

DEFAULTS:
          -t=2   # 2 simulation runs
          -p=1   # 1 simulation at-a-time

EXAMPLE:
          Run 10 simulations, 4 at-a-time, straight.xml mission, 
          with test-1.xml ranges file:
              ./scrimmage_runner.sh -t 10 -p 4 -m ~/scrimmage/scrimmage/missions/straight.xml \
              -r ~/scrimmage/scrimmage/config/ranges/test-1.xml

savasturk, 2018-07-06:
Thanks, I applied all of them but I am receiving error like this:

Traceback (most recent call last):
  File "generate_scenarios.py", line 47, in <module>
    import scrimmage.utils as utils
  File "/home/pinar/scrimmage/scrimmage/python/scrimmage/__init__.py", line 33, in <module>
    from scrimmage.utils import *
  File "/home/pinar/scrimmage/scrimmage/python/scrimmage/utils.py", line 44, in <module>
    import google.protobuf.internal.decoder
ModuleNotFoundError: No module named 'google'
Failed to generate scenarios
 
then run "sudo pip install conda" and "pip install google " but couldn't solve this error.

savasturk, 2018-07-09:
Output like:
pip 10.0.1
Python 3.6.3
you are right,how can I solve this probem?
Thanks a lot .

esquires, 2018-07-09:
My guess is that you just need to specify pip3. To make really sure everything is the right version though, try removing your build directory and executing the following

```[bash]
rm CMakeCache.txt
cmake .. -DPYTHON_MIN_VERSION=3.0
make

sudo pip3 install protobuf==3.3.0 grpcio==1.2.1 #note the 3
cd /path/to/scrimmage/python
sudo pip3 install -e .  # note the 3

```

savasturk, 2018-07-10:
I tried them but when run ":~/scrimmage/scrimmage/scripts$ ./scrimmage_runner.sh -t 100 -m ../missions/straight-vs-motorschemas.xml -p 7 " in this directory, Ireceived same error. These solutions couldn't solve my error.

esquires, 2018-07-10:
Perhaps if we simplify the setup we can identify the issue. Let's ignore generate_scenarios.py and scrimmage-runner.sh for a moment. Does this work?

```[bash]
$ python3
Python 3.5.2 (default, Nov 23 2017, 16:37:01)
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import google.protobuf.internal.decoder

```

savasturk, 2018-07-11:
Yes @esquires  it was work.

~/scrimmage/scrimmage/scripts$ python3
Python 3.6.3 |Anaconda custom (64-bit)| (default, Oct 13 2017, 12:02:49) 
[GCC 7.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import google.protobuf.internal.decoder
>>>

esquires, 2018-07-12:
Can you post the following?

`which python`

In build/CMakeCache.txt, the line

`PYTHON_EXECUTABLE:FILEPATH=/usr/bin/python2.7`

esquires, 2018-07-17:
I am going to go ahead and close this. If this is still an issue feel free to re-open and we'll help to resolve your issue.

closed 2018-07-17 by esquires