gtri/scrimmage #103: Create an Autonomy Plugin

savasturk, 2018-07-10:
Hi I'm following this tutorial (http://www.scrimmagesim.org/docs/sphinx/html/tutorials/autonomy-plugin.html ) and run these commands. But when run "make" command I received this error.

~/scrimmage/my-scrimmage-plugins/build$ make
[ 25%] Building CXX object src/plugins/autonomy/ExamplePlugin/CMakeFiles/ExamplePlugin_plugin.dir/ExamplePlugin.cpp.o
In file included from /opt/scrimmage/x86_64-linux-gnu/include/scrimmage/autonomy/Autonomy.h:37:0,
                 from /home/pinar/scrimmage/my-scrimmage-plugins/include/my-scrimmage-plugins/plugins/autonomy/ExamplePlugin/ExamplePlugin.h:35,
                 from /home/pinar/scrimmage/my-scrimmage-plugins/src/plugins/autonomy/ExamplePlugin/ExamplePlugin.cpp:33:
/opt/scrimmage/x86_64-linux-gnu/include/scrimmage/plugin_manager/Plugin.h:36:23: fatal error: Eigen/Dense: No such file or directory
compilation terminated.
src/plugins/autonomy/ExamplePlugin/CMakeFiles/ExamplePlugin_plugin.dir/build.make:62: recipe for target 'src/plugins/autonomy/ExamplePlugin/CMakeFiles/ExamplePlugin_plugin.dir/ExamplePlugin.cpp.o' failed
make[2]: *** [src/plugins/autonomy/ExamplePlugin/CMakeFiles/ExamplePlugin_plugin.dir/ExamplePlugin.cpp.o] Error 1
CMakeFiles/Makefile2:207: recipe for target 'src/plugins/autonomy/ExamplePlugin/CMakeFiles/ExamplePlugin_plugin.dir/all' failed
make[1]: *** [src/plugins/autonomy/ExamplePlugin/CMakeFiles/ExamplePlugin_plugin.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

I use 16.04 LTS Ubuntu dual boot.
pip 10.0.1
Python 3.6.3

How can I solve this error?
Thanks.

esquires, 2018-07-10:
It looks like the tutorial was out of date so I updated it. When the documentation job completes it will show up on scrimmagesim.org. In the meantime, you can build it locally with:

```[bash]
cd build
cmake .. -DBUILD_DOCS=ON
make docs
firefox docs/sphinx/html/index.html
```

That said, the error you got seems independent on the tutorial. For some reason it is not finding eigen. Can you open ``scrimmage/build/scrimmage-targets.cmake`` has something like this in it?

```
set_target_properties(scrimmage-core PROPERTIES
  INTERFACE_INCLUDE_DIRECTORIES "/path/to/scrimmage/include;/usr/include/eigen3;/usr/include"
  INTERFACE_LINK_LIBRARIES "scrimmage-protos;scrimmage-msgs;/usr/lib/x86_64-linux-gnu/libGeographic.so;dl"
)
```

I am looking in particular for the bit about ``/usr/include/eigen3``. If it is not there and you want a quick fix, you can put this in ``my-scrimmage-plugins/CMakeLists.txt``

```
find_package(Eigen3 QUIET)
include_directories(${EIGEN3_INCLUDE_DIR})
```

crichardson332, 2018-07-10:
A new pull request has been created to potentially solve this issue: pull request #104. Once this is merged, go ahead and update your scrimmage fork, re- build and install scrimmage, and then try building your plugin project again.

esquires, 2018-07-10:
@savasturk, you can test the fix as follows:

```[bash]

# get the pull request info
cd scrimmage # where your repo exists
git remote add crichardson332 https://github.com/crichardson332/scrimmage.git
git fetch crichardson332
git checkout -b crichardson332_master crichardson332/master

# start from scratch (or you can just remove build/CMakeCache.txt if you prefer to skip this section)
rm -rf build
mkdir build
cd build
cmake ..
make

cd ~/my-scrimmage-plugins
rm -rf build
mkdir build
cd build
cmake ..
make
```

esquires, 2018-07-10:
By the way, I noticed that you have installed scrimmage under ``/opt``. This should be fine, and the pull request should fix it, but in general you don't need to install scrimmage to use it with other plugins. If you want to remove the installation, you can do (before removing the build directory above)

``[bash]
cd scrimmage/build
cat install_manifest.txt| sudo xargs rm -rfv
``

savasturk, 2018-07-11:
I applied all of them, but they couldn't solve this error.

crichardson332, 2018-07-11:
Could you post the contents of the file `/opt/scrimmage/x86_64-linux-gnu/lib/scrimmage/cmake/scrimmage-targets.cmake`, specifically the following function:

` set_target_properties(scrimmage-core PROPERTIES ... )`

esquires, 2018-07-17:
I am going to go ahead and close this. If this is still an issue feel free to re-open and we'll help to resolve your issue.

closed 2018-07-17 by esquires