gtri/scrimmage #110: OpenAI Plugin Error

slickboy, 2018-07-16:
Hi, I was going through the OpenAI Plugin Tutorial and everything ran/built correctly until I ran `test_openai.py`. I received the following error

> Traceback (most recent call last):
  File "test_openai.py", line 34, in <module>
    test_openai()
  File "test_openai.py", line 18, in test_openai
    env = gym.make('scrimmage-v0')
  File "/home/rajan/.local/lib/python2.7/site-packages/gym/envs/registration.py", line 167, in make
    return registry.make(id)
  File "/home/rajan/.local/lib/python2.7/site-packages/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/home/rajan/.local/lib/python2.7/site-packages/gym/envs/registration.py", line 86, in make
    env = cls(**self._kwargs)
IndexError: map::at

Please let me know how I can resolve this error.

slickboy, 2018-07-20:
Thanks for the help @esquires ! I actually managed to resolve the error- in the mission xml file, I had to pass in the "radius" parameter. 
`<autonomy radius="1.0">SimpleLearner</autonomy>`

esquires, 2018-07-23:
Thanks. scrimmage now has updated documentation to reflect that suggestion.

http://www.scrimmagesim.org/docs/sphinx/html/tutorials/openai-plugin.html#plugin-parameter-file

It looks like you re-opened the ticket. Do you need anything else?