gtri/scrimmage #114: OpenAI Continuous Action Issue

slickboy, 2018-07-20:
Hi, one more issue with the OpenAI plugin. I tried to add 2 ranges to the continuous_extrema list: 

`
   action_space.continuous_extrema.push_back(std::make_pair(-inf,inf));
   action_space.continuous_extrema.push_back(std::make_pair(-inf,inf));
`
When I run openai.py, I get the following error: 

>   File "test_openai.py", line 46, in <module>
    test_openai()
  File "test_openai.py", line 18, in test_openai
    env = gym.make('scrimmage-v0')
  File "/home/rajan/.local/lib/python2.7/site-packages/gym/envs/registration.py", line 167, in make
    return registry.make(id)
  File "/home/rajan/.local/lib/python2.7/site-packages/gym/envs/registration.py", line 130, in make
    max_episode_seconds=env.spec.max_episode_seconds)
  File "/home/rajan/.local/lib/python2.7/site-packages/gym/wrappers/time_limit.py", line 6, in __init__
    super(TimeLimit, self).__init__(env)
  File "/home/rajan/.local/lib/python2.7/site-packages/gym/core.py", line 243, in __init__
    self.action_space = self.env.action_space
TypeError: Unable to convert function return value to a Python type! The signature was
	(self: scrimmage.bindings.py_bindings.ScrimmageOpenAIEnv) -> object

esquires, 2018-07-23:
Thanks for filing a ticket. This looks like a bug. When creating a tuple space in openai we need to pass a tuple of spaces rather than individual arguments. I should have a fix in by the end of the day.

esquires, 2018-07-23:
This should now be fixed in the latest master (2d2fa32). I added a couple test cases in ``test_openai.py`` to test for the use of a tuple space. See the ``test_two_dim_tuple`` function.

Note that in addition to adding the lines

```[c++]
action_space.continuous_extrema.push_back(std::make_pair(-inf,inf)); 
action_space.continuous_extrema.push_back(std::make_pair(-inf,inf));
```

you will also need to update your action to be multi-dimensional (I am assuming you kept the discrete action input). Let me know if you are still having difficulties.

slickboy, 2018-07-23:
Thanks for the fix! I'm still a bit confused about how I actually pass in a multi-dimensional continuous action for a multi-agent scenario. Could you point me to an example?

esquires, 2018-07-23:
Sure. 

https://github.com/gtri/scrimmage/blob/master/test/test_openai.py#L167

The mission file that is used by that test is pasted below:

```[xml]
<runscript name="Straight flying">

  <run dt="1" enable_gui="true" end="1000" network_gui="false" start="0.0" start_paused="true" time_warp="0" />

  <stream_port>50051</stream_port>
  <stream_ip>localhost</stream_ip>

  <end_condition>time</end_condition> 

  <grid_spacing>1</grid_spacing>
  <grid_size>1000</grid_size>

  <background_color>191 191 191</background_color> 
  <gui_update_period>10</gui_update_period> 

  <plot_tracks>false</plot_tracks>
  <output_type>nothing</output_type>
  <display_progress>false</display_progress>
  <show_plugins>false</show_plugins>

  <log_dir>~/.scrimmage/logs</log_dir>

  <latitude_origin>35.721025</latitude_origin>
  <longitude_origin>-120.767925</longitude_origin>
  <altitude_origin>300</altitude_origin>
  <show_origin>false</show_origin>
  <origin_length>10</origin_length>
  <camera focal_point="0,0,0" follow_id="4" mode="free" pos="0,0.01,50" show_fps="false" />

  
  
  

  <seed>3361078855</seed>

  <entity_common name="all">
      <count>1</count>
      <health>1</health>
      <radius>1</radius>

      <team_id>1</team_id>
      <visual_model>Sphere</visual_model>
      <motion_model>SingleIntegrator</motion_model>
      <controller>SingleIntegratorControllerSimple</controller>
      <sensor order="0">RLSimpleSensor</sensor>
      <autonomy ctrl_y="True" discrete_x="true" discrete_y="true" x_discrete="False" y_discrete="True">RLSimple</autonomy>
      <y>0</y>
      <z>0</z>
  </entity_common>

  <entity entity_common="all">
    <x>0</x>
    <color>77 77 255</color>
  </entity>
</runscript>

```

slickboy, 2018-07-24:
It seems this is for 1 agent (<count>1</count>)- I'm a bit confused about how to set the python and mission file up for a team of agents and pass in a multidim, continuous action for each agent from openai. Just for background, I am doing work on neuroevolution with homogeneous teams of robots.

slickboy, 2018-07-24:
It seems this is for 1 agent (`<count> 1 </count>`) I'm a bit confused about how to set the python and mission file up for a team of agents and pass in a multidim, continuous action for each agent from openai. Just for background, I am doing work on neuroevolution with homogeneous teams of robots.

esquires, 2018-07-24:
My apologies. I thought you were trying to have a discrete and continuous action for one agent.

There are 3 test cases shown in [test_openai.py](https://github.com/gtri/scrimmage/blob/master/test/test_openai.py#L207) reflecting the basic use cases of multiple agents. In general, for multiple agents, you have the option of viewing them on the python side as a single agent (a centralized approach) or as individual agents. You can toggle this option with ``combine_actors``. Additionally, if you want to set ``combine_actors`` to true and the agents all have the same global sensor, you can set ``global_sensor`` to true so that the state is not repeated needlessly.

The 3 test cases are:

* test_two_combined_veh_dim_discrete - Two agents, each with one discrete input, treated as a single agent.

* test_two_not_combined_veh_dim_discrete - Two agents, each with discrete input, treated as separate agents.

* test_two_combined_veh_dim_discrete_global_sensor - Two agents, each with discrete input, treated as a single agent.

Let me know if you need additional clarification.