gtri/scrimmage #115: Mission <start_paused> Tag

lvalcourt, 2018-07-23:
The <start_paused> tag will not work. I tried changing the values inside the xml file and inside a function in a python file that recreated the xml with new tags. Setting it to true or false, it will always starts as paused.

SyllogismRXS, 2018-07-23:
The "start_paused" variable is an attribute of the <run/> tag in your main mission file. It seems to be working correctly in the ./missions/straight.xml example file. Are you using the start_paused variable like this?

`<run start="0.0" end="100" dt="0.1"
       time_warp="10"
       enable_gui="true"
       network_gui="false"
       start_paused="true"/>`

lvalcourt, 2018-07-23:
Yes, I am using start_paused like that
This is my <run> tag
<run start="0.0" end="100" dt="1" time_warp="1" enable_gui="true" network_gui="false" start_paused="true", num_actors="3"/>

However, I use the _write_temp_file function used in test_openai.py to override the file in case I want to use more actors or change discrete values. 

Thanks!