gtri/scrimmage #160: Waypoint requires WaypointGenerator_plugin to be linked in CMake

gtcooke94, 2018-10-17:
With a recent change to Waypoint, we have to explicit link WaypointGenerator_plugin in CMake to use the Waypoint class at all:
https://github.com/gtri/scrimmage/blob/b494d62271b80fbc4e8b5727e4de85a4bc3c3859/src/plugins/autonomy/MoveToGoalMS/CMakeLists.txt#L15-L18
I have been using this class outside of times where I use WaypointGenerator. I think Waypoint is the kind of thing that could move to common? That way it's also not under the scrimmage::autonomy namespace, which I don't necessarily think is an appropriate place for Waypoint.
@esquires it seems you've been working with this recently too, what do you think?

esquires, 2018-10-17:
This issue has come up in other contexts where a plugin has a set of functionality that would be useful elsewhere even if linking to the actual plugin isn't necessary. There seems to be a few options here:

1. Move library functionality into headers (the way Waypoint was previously). This will have a negative impact on build times.
2. Link to the library. This may create installation issues since the plugin libraries are not placed under `prefix/lib`
3. Create a utilities library in the plugin CMakeLists file that separate from the plugin and link the plugin shared object to the utilities library. On install, put the utilities library in the same place as scrimmage-core
4. Put stuff into scrimmage-core. This will likely mean we put stuff peripheral stuff in scrimmage-core.

I am inclined to go with (3). @SyllogismRXS , what do you think?

crichardson332, 2018-11-29:
This issue has been resolved with https://github.com/gtri/scrimmage/pull/184. `Waypoint` is now part of the `scrimmage-core` library.

esquires, 2018-11-29:
Thanks. closing.

closed 2018-11-29 by esquires