gtri/scrimmage #231: Possible bug in straight.xml mission file

asydney, 2019-01-31:
Answer the following questions:

* what are you trying to do?
Run the following mission:
$scrimmage missions/straight.xml

* what is the problem and how can it be recreated?
I get following error:
"""
terminate called after throwing an instance of 'std::invalid_argument'
  what():  stoi
Aborted (core dumped)
"""
* what scrimmage commit are you on? You can see this with `git rev-parse HEAD`
e2b2026575f155df0b14ab9afdb0d6e17ac3302c

* describe any changes you made to scrimmage.
So this worked fine back in the day(say late last year) until I did a git pull a few days ago. I've done a diff below to show the issue:

61c61
<     <count>${count=1}</count>
---
>     <count>1</count>

Changing line 61 to "<count>1</count>" resolves the problem.

SyllogismRXS, 2019-01-31:
It may be that the "scrimmage" binary that you are running isn't the one that you are building. Can you run "which scrimmage" and make sure it is the one located in your build tree at "/path/to/scrimmage/build/bin/scrimmage". If you have a very old version of scrimmage, it may be worth updating the dependencies and rebuilding scrimmage again. Here are the steps:

1. Remove your scrimmage build directory: $ rm -rf /path/to/scrimmage/build
2. Remove your scrimmage config directory: $ rm -rf ~/.scrimmage
3. Update your scrimmage dependencies: 
      $ sudo apt-get update
      $ sudo apt-get install scrimmage-dependencies scrimmage-pybind11 \
         scrimmage-protobuf3 scrimmage-grpc scrimmage-jsbsim
4. Rebuild scrimmage: 
    $ source /opt/scrimmage/*/setup.sh
    $ cd /path/to/scrimmage
    $ mkdir build && cd build
    $ cmake .. && make -j7

Let me know if any of those steps don't work.

asydney, 2019-02-01:
I got the following errors when I attempted to run the mission staight.xml

> 

Failed to find configuration: mcmillan
Failed to find configuration: GlobalNetwork
Failed to parse: GlobalNetwork.xml for type scrimmage::Network
Check that you have sourced ~/.scrimmage/setup.bash (this sets the environment variable SCRIMMAGE_PLUGIN_PATH to a directory including GlobalNetwork.xml)

But everything now works after doing the following (i.e. as suggested above):

> 

source ~/.scrimmage/setup.bash

So I'm all set. Thanks a bunch!

closed 2019-02-01 by asydney