gtri/scrimmage #240: Aircraft disappear when motion_multiplier=1

crebhuhn, 2019-02-06:
Perhaps more documentation is needed on 'motion_multiplier' in the xml run tag, but when I set it to 1 (as instructed https://gtri.github.io/scrimmage/sphinx/html/overview/xml-tags.html) additional entities that were spawned by the 'count' tag seem to disappear.

This bug can be re-created by setting <run motion_multiplier="1"> at the top of the missions/predator_prey_boids.xml file.

I am on scrimmage commit 1e816a626e2de709c23ce7f4ba12a2b7f9424925 and have not modified scrimmage (only developed plugins in a separate directory from the scrimmage base).

SyllogismRXS, 2019-02-12:
The problem here is that the JSBSimModel motion model requires a small time step (dt) during it's ODE integration. If the time step is too large for a motion model that is highly dynamic, then the numerical approximation methods diverge, which fills the motion model state with NaN values. When the NaN values are received, the entity is basically removed from the simulation. I will look into adding some error checking to at least print out that the entity has NaNs in its state. You can swap out the JSBSimModel with a simpler motion model and set the motion_multiplier to '1', which works fine. For example, in predator_pre_boids.xml, I commented out the JSBSimModel portion of the first entity:

Add added the Unicycle model instead...

    <motion_model pitch_rate_max="1.5" turn_rate_max="2" vel_max="35">Unicycle</motion_model>
    <controller>DirectController</controller>

You can then set the motion_multiplier to '1' and the vehicle states don't go to NaN.

crebhuhn, 2019-02-12:
Thank you for your explanation Kevin, that does make sense. I definitely agree that error checking (or noisy failing) would help diagnose this problem faster next time, especially if bounds on the inputs like motion_multiplier and dt are not well-defined for convergence of numerical approximators.

SyllogismRXS, 2019-02-13:
The modification to SimControl.cpp in pull request #253 prints warnings to the screen if the entity's position contains a NaN value. The entity is then removed, but the rest of the simulation is allowed to continue. Interestingly, in the predator_prey_boids.xml mission, some JSBSimModel motion models contain NaN values even if the motion_multiplier isn't set to '1'. This will require more debugging and is probably an issue with JSBSim, but I think this specific issue can be closed.

closed 2019-02-13 by SyllogismRXS