gtri/scrimmage #282: GPRC issue running tensorflow based code

asydney, 2019-03-28:
Answer the following questions:

* what are you trying to do?
I'm trying to run the "Create an OpenAI Plugin": essentially use tensorflow with scrimmage

* what is the problem and how can it be recreated?
  - I've got the tensorflow code in ~/scrimmage/syd-plugins/my_openai.py
  - Per instructions from section "Run In Non-learning..." of the tutorial, I have updated SimpleLearner.xml as follows:
``
<params>
  <library>SimpleLearner_plugin</library>
  <radius>2</radius>
  <module>my_openai</module>
  <actor_func>get_action</actor_func>
  <grpc_mode>true</grpc_mode>
  <grpc_address>localhost</grpc_address>
  <port>50051</port>
</params>
``

  - Below is the error:

``
$ scrimmage missions/openai_mission.xml 
sh: 1: openai_grpc_link.py: not found
Connecting to GRPC Server
Attempt 1/ 10
Can't Connect: message 14, Connect Failed
Attempt 2/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 3/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 4/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 5/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 6/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 7/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 8/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 9/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 10/ 10
Can't Connect: message 4, Deadline Exceeded
Didn't connect to the grpc server. Make sure the grpc client is running and your python bindings are up to date.
failed to update entity 1, plugin type "Autonomy", plugin name "SimpleLearner"
Exiting due to plugin request.

Simulation Complete

``

* what scrimmage commit are you on? You can see this with `git rev-parse HEAD`
39cc850dd1b04fbcc54281bb3c082defb6091092

* describe any changes you made to scrimmage.
None. Just followed the instructions from the tutorial mentioned above

* If you can recreate the issue using only plugins in the scrimmage repository, what mission file are you running?
This is mentioned above.

bogidude, 2019-03-29:
From this line in your output:
```
sh: 1: openai_grpc_link.py: not found
```
There could be a couple of possibilities of what is going wrong. 

1) Can you make sure that the `openai_grpc_link.py` from python/scrimmage was installed properly? That can be done be running the following command in the terminal:
```
which openai_grpc_link.py
```
That should return a file location to where `openai_grpc_link.py` was installed on your system. If it returns nothing, then you need to reinstall the python bindings following the instructions under "Install SCRIMMAGE Python Bindings" on the README (located [here](https://github.com/gtri/scrimmage/blob/master/README.md#install-scrimmage-python-bindings)).

2) The other potential problem is python not knowing where your `~/scrimmage/syd-plugins/my_openai.py` file is located. You can check whether or not python knows to look for python files at that location by echoing the $PYTHONPATH variable. If `~/scrimage/syd-plugins` isn't there, it can be added to the PYTHONPATH environment variable, like so:
```
export PYTHONPATH=~/scrimmage/syd-plugins/:$PYTHONPATH
```
I don't think 2) will create that error output but I can't remember off of the top of my head. If neither of those work, post a followup comment.

asydney, 2019-04-01:
Let me provide some context. Initially, I installed scrimmage in ~/git/scrimmage. Then per the instructions in "Create an OpenAI Plugin", I created a new scrimmage plugins project in ~/scrimmage. Here are a few more notes:
1. openai_grpc_link.py exists in the original ~/git/scrimmage/python/scrimmage, but not in ~/scrimmage/syd-plugins.
2. I copied ~/git/scrimmage/python over to ~/scrimmage/syd-plugins and installed the python bindings as follows:
cd ~/scrimmage/syd-plugins/python
sudo pip install -e .
3. When I run scrimmage this time around, I get the following:
``
$ scrimmage missions/openai_mission.xml 
Traceback (most recent call last):
  File "/usr/local/bin/openai_grpc_link.py", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2927, in <module>
    @_call_aside
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2913, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2940, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 637, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 650, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 829, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'protobuf==3.3.0' distribution was not found and is required by scrimmage
Connecting to GRPC Server
Attempt 1/ 10
Can't Connect: message 14, Connect Failed
Attempt 2/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 3/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 4/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 5/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 6/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 7/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 8/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 9/ 10
Can't Connect: message 4, Deadline Exceeded
Attempt 10/ 10
Can't Connect: message 4, Deadline Exceeded
Didn't connect to the grpc server. Make sure the grpc client is running and your python bindings are up to date.
failed to update entity 1, plugin type "Autonomy", plugin name "SimpleLearner"
Exiting due to plugin request.

Simulation Complete
``
Note: It appears the scrimmage is using python2.7 but tensorflow (and gym) require python3. As a result I've installed them as follows:

python3.5 -m pip install --user --upgrade pip==9.0.3
sudo pip3 install tensorflow
sudo pip3 install --upgrade gym

bogidude, 2019-04-03:
You shouldn't need to copy the ``openai_grpc_link.py`` to your ``~/scrimmage/syd-plugins``. Your system just needed to know where it is located, which is done by running the ``sudo pip install -e .`` command.

From the output you provided:
```
pkg_resources.DistributionNotFound: The 'protobuf==3.3.0' distribution was not found and is required by scrimmage
```
It looks like you are having problems with python package versions. In scrimmage's python package, we expect protobuf version 3.3.0. If you install the latest tensorflow, that will require a newer version of protobuf (I think 3.6.5). This can be fixed by either installing an older tensorflow that uses protobuf version 3.3.0 or just changing what scrimmage requires. Changing the protobuf version scrimmage python requires shouldn't break anything so I would recommend changing what scrimmage requires by doing the following:

Open up ``~/git/scrimmage/CMakeLists.txt`` in your favorite text editor. Find the line
```
set(PROTOBUF_VERSION 3.3.0)
```
and change ``3.3.0`` to match your python3's protobuf version number. This CMake variable is used in ``~/git/scrimmage/python/setup.py.in`` as the version of protobuf required for the scrimmage python package.  After you change it to the correct version number, you can do the following:

```
$ cd ~/git/scrimmage/build
$ cmake ..
```
You can double check that it changed properly by opening up the ``~/git/scrimmage/python/setup.py`` file and seeing what version of protobuf it requires. Then reinstall the scrimmage python package as follows (use ``pip3`` instead of ``pip`` below to install it for python3):
```
$ cd ~/git/scrimmage/python
$ sudo pip install -e .
```

As for which python scrimmage is using, it by default will use python2.7 like you said. If you want to change it to use python3.5, you will need to do the following in the terminal:
```
$ rm  -rf ~/git/scrimmage/build/*
$ cd ~/git/scrimmage/build
$ cmake .. -DPYTHON_EXECUTABLE=$(which python3.5)
$ make
```

That will remove everything in your build folder including all cached CMake references to python2,7. Then it recreates the makefile using the location of the python3 executable, provided by ``$(which python3.5)``. You can also type out the path to the python3 executable yourself if you want instead of using ``$(which python3.5)``. Then when you rebuild scrimmage, it should be built using python3. At this point, I don't think you will need to rerun the ``sudo pip3 install -e .`` command from ``~/git/scrimmage/python`` but if something fails, try running it again just to be safe. Hope this helps!

asydney, 2019-04-09:
So I did the following"
1. Worked with python3.5.2
2.  Changed protobuf version to what will be used by tensorflow 1.5 and build SCRIMMAGE Core:
> 
vim machineLearning/git/scrimmage/CMakeLists.txt (line 145)
...
set(PROTOBUF_VERSION 3.4.0)
...
cd ~/git/scrimmage/build
source ~/.scrimmage/setup.bash
cmake .. -DPYTHON_EXECUTABLE=$(which python3)
make
> 

3. Used pip3 to install tensorflow 1.5 (which ships with protobuf 3.4.0). 
4. Install python bindings:
> 
pip3 install protobuf==3.4.0 grpcio==1.2.1
cd ~/git/scrimmage/python
pip3 install -e .
> 

and scrimmage seems happy at the moment. Thanks!

closed 2019-04-09 by asydney