gtri/scrimmage #291: Scrimmage dependencies

dr-parker, 2019-04-14:
I'm trying to install Scrimmage, but after executing this line:

'sudo add-apt-repository ppa:kevin-demarco/scrimmage'

Received the following:

 SCRIMMAGE multi-agent robotics simulator.
 More info: https://launchpad.net/~kevin-demarco/+archive/ubuntu/scrimmage
Press [ENTER] to continue or Ctrl-c to cancel adding it.
Hit:1 http://security.ubuntu.com/ubuntu bionic-security InRelease
Hit:2 http://us.archive.ubuntu.com/ubuntu bionic InRelease
Hit:3 http://us.archive.ubuntu.com/ubuntu bionic-updates InRelease
Hit:4 http://packages.ros.org/ros/ubuntu bionic InRelease
Hit:5 http://us.archive.ubuntu.com/ubuntu bionic-backports InRelease
Ign:6 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu bionic InRelease
Err:7 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu bionic Release
  404  Not Found [IP: 91.189.95.83 80]
Hit:8 https://packagecloud.io/slacktechnologies/slack/debian jessie InRelease
Reading package lists...

The repository 'http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Any insights would be much appreciated so I could continue the install. Thanks!

SyllogismRXS, 2019-04-15:
It looks like you are using Ubuntu 18.04 (bionic), right? Our current installation instructions are for Ubuntu 16.04, but we do support 18.04 as well. You just need to add the PPA for scrimmage-bionic instead. First, remove the scrimmage PPA for 16.04:

$ sudo add-apt-repository --remove ppa:kevin-demarco/scrimmage

Now, you can add the scrimmage 18.04 (bionic) PPA:

$ sudo add-apt-repository ppa:kevin-demarco/scrimmage-bionic
$ sudo apt-get update

The scrimmage 18.04 dependencies are also located on launchpad:
https://launchpad.net/~kevin-demarco/+archive/ubuntu/scrimmage-bionic

I'll update the README.md instructions to note the difference for 18.04 installations. Let me know if you have any problems.

dr-parker, 2019-04-15:
Awesome! Will do, thanks so much for the quick turn-around. Will keep you
posted.

Best,
Lonnie

dr-parker, 2019-04-15:
Just following up that that worked! Was able to run the straight.xml example. 

Discovered additional issues, but fixed independently, just adding below for reference in case anyone else comes across them:

1) Upon initial 'make' command, ran into a fatal error for rapidxml: rapdixml/rapidxml.hpp not found. Once up upgraded for Ubuntu 18.04 Bionic, with sudo apt-get install librapidxml-dev, the make completed fine.

2) Noticed after cmake command, a message/warning re: VTK. I've attached the terminal output for reference. This did not impact the successful make, but thought the message was worth taking note of. I've attached the output for reference.
[vtk-minor.txt](https://github.com/gtri/scrimmage/files/3078678/vtk-minor.txt)

closed 2019-04-15 by dr-parker