gtri/scrimmage #311: Issues in an attempt to control a drone via keyboard/python script

asydney, 2019-05-13:
PS. I'm not sure if this is the correct forum to ask the following, but I'll go ahead and place it there and please provide feedback accordingly. 

Answer the following questions:

* what are you trying to do?
- I have re-purposed fixed_wing_6dof.xml and I would like to create a mission such that I've got 1 aircraft flying in a circle of a given radius(or some random pattern within a map), and I would like the ability to control a second aircraft (via a. the keyboard, and b. commands that can be passed in a python script) on the same map. 
- Based on conversations with Eric, I did the following for the first air-craft:

`<roll>25</roll>
    <script_name>rascal_piedmont.xml</script_name>
    <controller use_roll="true">JSBSimControlControllerHeadingPID</controller>
    <motion_model write_csv="true">JSBSimControl</motion_model>
`

* what is the problem and how can it be recreated?
a)  When I run the simulation, I see a series of red/blue lines: I've uploaded a snapshot at the link below:
https://pasteboard.co/IezcEYd.png

How do I get rid of these lines?

b) Eric mentioned that when I set the controller to expect roll, I need to output it to the autonomy. What exactly does this mean?  Should I specify an autonomy plugin(i.e. in <autonomy></autonomy>)? If yes, how do I determine which one to use? For example, I see options for "Straight, JoystickAutonomy etc"

c) How do I go about adding the keyboard functionality to control a drone in the simulation(and control as in speed, altitude, heading etc.)? 

d) How would I go about specifying the actions to control a drone in a python script (as opposed to a keyboard)?

* what scrimmage commit are you on? You can see this with `git rev-parse HEAD`
 - 67d5ac9a2b383d244c1967d72360ab4222e39bc5

* describe any changes you made to scrimmage.
- N/A

* If you can recreate the issue using only plugins in the scrimmage repository, what mission file are you running?
- N/A

FX196, 2019-06-27:
Could you share how you were able to connect scrimmage to Python? I'm trying to implement keyboard inputs and none of the python bindings are working for me.

asydney, 2019-06-27:
Unfortunately, I have not been able to get this to work.

-Ali

shaun-d-anderson, 2020-12-01:
a) JSBSimControl is drawing Velocity, Angular Velocity and Acceleration vectors by default.  These can be set in the JSBSimControl.xml file as    <drawVel>0</drawVel>,  <drawAngVel>0</drawAngVel>, <drawAcc>0</drawAcc>
b) An autonomy will need to publish to the vehicle controller the desired path, heading, roll, etc. depending on the controller input
c) Take a look at the joystick controller plugin for external control of the aircraft.
d) A python autonomy (scripted actions) can be developed and integrated.  Scrimmage and ROS can be utilized to ease integration with python autonomy execution.

closed 2020-12-01 by shaun-d-anderson