gtri/scrimmage #319: CV_BGR2GRAY and CV_LOAD_IMAGE_COLOR removed in latest OpenCV

palbers, 2019-05-29:
When updating to the new scrimmage with the ground collision plugin, I get a compile error:

/scrimmage/src/plugins/interaction/MapGen2D/MapGen2D.cpp:115:37: error: use of undeclared identifier 'CV_LOAD_IMAGE_COLOR'
    map_img_ = cv::imread(filename, CV_LOAD_IMAGE_COLOR);
                                    ^
/scrimmage/src/plugins/interaction/MapGen2D/MapGen2D.cpp:180:29: error: use of undeclared identifier 'CV_BGR2GRAY'
    cv::cvtColor(img, gray, CV_BGR2GRAY);
                            ^
2 errors generated.

after some investigation, I uncovered this error:

$ cat imgcodecs_c.h
#error "This header with legacy C API declarations has been removed from OpenCV. Legacy contants are available from legacy/constants_c.h file."

richardnnn, 2019-06-14:
This is due to OpenCv version issue, replacing following deprecated constants make code recompile for me:
Inside scrimmage/src/plugins/interaction/MapGen2D/MapGen2D.cpp, 
Line 115: change “CV_LOAD_IMAGE_COLOR” to “cv::IMREAD_COLOR” 
Line 180: Change “CV_BGR2GRAY” to “cv::COLOR_BGRA2GRAY”

SyllogismRXS, 2019-06-18:
Thanks for the fix. I just submitted a pull request to incorporate these changes. When it passes the Travis-CI build/tests, I'll merge it into master: https://github.com/gtri/scrimmage/pull/333

SyllogismRXS, 2019-06-18:
Resolved through https://github.com/gtri/scrimmage/pull/333

closed 2019-06-18 by SyllogismRXS