gtri/scrimmage #330: Failed to find configuration when running Mission File

Yibin-Li, 2019-06-15:
When I follow the installation instruction and run the scrimmage ./missions/straight.xml, I have an error 

Failed to find configuration: mcmillan
Failed to find configuration: GlobalNetwork
Failed to parse: GlobalNetwork.xml for type scrimmage::Network
Check that you have sourced ~/.scrimmage/setup.bash (this sets the environment variable SCRIMMAGE_PLUGIN_PATH to a directory including GlobalNetwork.xml)
Failed to find configuration: LocalNetwork
Failed to parse: LocalNetwork.xml for type scrimmage::Network
Check that you have sourced ~/.scrimmage/setup.bash (this sets the environment variable SCRIMMAGE_PLUGIN_PATH to a directory including LocalNetwork.xml)
Failed to find configuration: SimpleCollisionMetrics
Failed to parse: SimpleCollisionMetrics.xml for type scrimmage::Metrics
Check that you have sourced ~/.scrimmage/setup.bash (this sets the environment variable SCRIMMAGE_PLUGIN_PATH to a directory including SimpleCollisionMetrics.xml)
Failed to find configuration: Boundary
Failed to parse: Boundary.xml for type scrimmage::EntityInteraction
Check that you have sourced ~/.scrimmage/setup.bash (this sets the environment variable SCRIMMAGE_PLUGIN_PATH to a directory including Boundary.xml)
Failed to find configuration: SimpleCollision
Failed to parse: SimpleCollision.xml for type scrimmage::EntityInteraction
Check that you have sourced ~/.scrimmage/setup.bash (this sets the environment variable SCRIMMAGE_PLUGIN_PATH to a directory including SimpleCollision.xml)
Failed to find configuration: GroundCollision
Failed to parse: GroundCollision.xml for type scrimmage::EntityInteraction
Check that you have sourced ~/.scrimmage/setup.bash (this sets the environment variable SCRIMMAGE_PLUGIN_PATH to a directory including GroundCollision.xml)
WARNING: Plugin - Publisher unable to connect to network (GlobalNetwork) on topic (EndTime)
WARNING: Subscriber unable to connect to network (GlobalNetwork) on topic (GenerateEntity)
Failed to find configuration: zephyr-blue
Failed to find configuration: SimpleAircraft
Failed to parse: SimpleAircraft.xml for type scrimmage::MotionModel
Check that you have sourced ~/.scrimmage/setup.bash (this sets the environment variable SCRIMMAGE_PLUGIN_PATH to a directory including SimpleAircraft.xml)
Failed to parse entity at start position: x=-900, y=0
Failed to generate entity

I have run the `source ~/.scrimmage/setup.bash` command multiple times but it is still not working. I am using Ubuntu 16.04.

SyllogismRXS, 2019-06-18:
After you source the ~/.scrimmage/setup.bash file, what is the output of "echo $SCRIMMAGE_PLUGIN_PATH" ? Did scrimmage complete building successfully? (running the "make" command in the build directory).

Yibin-Li, 2019-06-18:
> After you source the ~/.scrimmage/setup.bash file, what is the output of "echo $SCRIMMAGE_PLUGIN_PATH" ? Did scrimmage complete building successfully? (running the "make" command in the build directory).

The build completed successfully. It looks that my computer doesn't run this bash file correctly. I end up writing a new shell script that exports all the environmental variables and source it. This shell script works and the simulation works as well.

closed 2019-06-18 by Yibin-Li

dr-parker, 2020-01-27:
In case it helps anyone else, I ran into the same issue and found that deleting the ~/scrimmage/build folder and re-building also worked. I ran into multiple issues (mainly attributed to being a noob), but zoned in on the fact that the setup.bash file cleanly manages the configuration (i.e. definition) of all the relevant Scrimmage environment variables (SCRIMMAGE_PLUGIN_PATH, SCRIMMAGE_CONFIG_PATH, SCRIMMAGE_DATA_PATH, PATH, and PYTHONPATH). While mucking around with the SCRIMMAGE_PLUGIN_PATH, that's where I encountered a lot of the above errors (e.g., "Fail to find *"). Can't say for sure what the exact action (or set of actions) was that got things working, but it was some combination of deleting the ~/scrimmage/build folder and re-building and running "source ~/.scrimmage/setup.bash".