gtri/scrimmage #335: Cannot locate PyAutonomy

richardnnn, 2019-06-20:
Answer the following questions:

* what are you trying to do?
Run `scrimmage missions/python-test.xml`

* what is the problem and how can it be recreated?
The following error message is returned:
```
/.../scrimmage/build/plugin_libs/libPyAutonomy_plugin.dylib doesn't contain 'plugin_type'
ERROR: Could not locate scrimmage::Autonomy::PyAutonomy_plugin
VariableIO Error: no autonomies provide inputs required by Controller "SimpleAircraftControllerPID". Add VariableIO output declarations in as follows
First, include the VariableIO class in the cpp file: #include <scrimmage/common/VariableIO.h>
Second, place the following in its initializer:
```
And when I do `scrimmage-plugin PyAutonomy`:
```
==========================================
Plugin found.
Name: PyAutonomy
File: /Users/nyfkxj/ucb_archive/swarm-on-swarm/repos/scrimmage_related/scrimmage/include/scrimmage/plugins/autonomy/PyAutonomy/PyAutonomy.xml
Library: PyAutonomy_plugin
-------------------------
Params: XML_DIR=/Users/nyfkxj/ucb_archive/swarm-on-swarm/repos/scrimmage_related/scrimmage/include/scrimmage/plugins/autonomy/PyAutonomy/
XML_FILENAME=/Users/nyfkxj/ucb_archive/swarm-on-swarm/repos/scrimmage_related/scrimmage/include/scrimmage/plugins/autonomy/PyAutonomy/PyAutonomy.xml
class=AuctionAssign
library=PyAutonomy_plugin
module=auction_assign
/Users/nyfkxj/ucb_archive/swarm-on-swarm/repos/scrimmage_related/scrimmage/build/plugin_libs/libPyAutonomy_plugin.dylib doesn't contain 'plugin_type'
---------------------------
Matching plugin libraries found:
```
* what scrimmage commit are you on? You can see this with `git rev-parse HEAD`
Newest, 92e3153e2022ccd259a09e8a1ac8ca56b4af5d4c

* describe any changes you made to scrimmage.
Nothing

* If you can recreate the issue using only plugins in the scrimmage repository, what mission file are you running?
python-test.xml

esquires, 2019-06-20:
Are you on OSX? We have only tested the python autonomy for Linux.

richardnnn, 2019-06-20:
Yes I’m on OSX.

aew61, 2019-07-03:
I am also having the exact same problem as above, but I am running Ubuntu 18 and python 3.6. Are there steps I can take to fix this?

esquires, 2019-07-04:
It looks like that may be an old mission file. Does rlsimple.xml work for you?

bobotran, 2019-10-31:
Is there a followup on this? I'm having the same issue.

TheDash, 2020-03-17:
Also having this issue

shaun-d-anderson, 2020-12-01:
With the addition of Variable-IO for controller and autonomy interaction the PyAutonomy plugin is deprecated and it is suggested to leverage the ROS integration which is supported in latest commits.

closed 2020-12-01 by shaun-d-anderson