gtri/scrimmage #342: Why two different PPAs for different versions?

flaviut, 2019-07-05:
The readme mentions adding the `ppa:kevin-demarco/scrimmage` PPA on xenial and the `ppa:kevin-demarco/scrimmage-bionic` PPA on bionic.

However, this doesn't match how most PPAs work, where packages for all OS versions are deployed to the same PPA. See https://launchpad.net/ubuntu/+ppas -- this specific PPA supports 12.04 through 19.10 in a single PPA.

SyllogismRXS, 2019-07-08:
Launchpad.net asked me if I wanted to migrate my xenial packages to bionic about a year ago and when I accepted the migration offer, it automatically created a separate PPA. I'll see if Launchpad will let me push both xenial and bionic packaes to a single PPA.

SyllogismRXS, 2019-10-31:
Both xenial and bionic are supported by the same launchpad PPA now. The README.md was updated to reflect this change. I'm going to leave the scrimmage-bionic PPA up for a couple of months and then take it down to avoid any confusion.