gtri/scrimmage #347: first time user trying to install, fails at 'cmake ..' line

davidbuzz, 2019-07-29:
Answer the following questions:

* what are you trying to do? 
install scrimmage according to github instructions

* what is the problem and how can it be recreated?
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find GeographicLib (missing: GeographicLib_INCLUDE_DIR
  GeographicLib_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindGeographicLib.cmake:15 (find_package_handle_standard_args)
  CMakeLists.txt:163 (find_package)

-- Configuring incomplete, errors occurred!

* what scrimmage commit are you on? You can see this with `git rev-parse HEAD`
git rev-parse HEAD
73300f36c85e4e5bcc27bbceaf70a280722cdac3

* describe any changes you made to scrimmage.
none.

SyllogismRXS, 2019-07-30:
Were you able to run the "install-binaries.sh" script successfully? This runs "apt-get install" on the required packages (geographiclib being one of them). https://github.com/gtri/scrimmage#install-binary-dependencies.

sudo apt-get install libgeographic-dev

davidbuzz, 2019-07-30:
install-binaries seemed to run ok, but looking back at the output it appears to have thrown some warnings....
Err:22 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu bionic Release  
  404  Not Found [IP: 91.189.95.83 80]
Reading package lists... Done                      
E: The repository 'http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
....
Package tcl-vtk is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'tcl-vtk' has no installation candidate

davidbuzz, 2019-07-30:
regarding the 'tcl-vtk' issue, on my machine the package is called 'tcl-vtk7' ( or  6) , no idea how to fix the missing Release file from the ppa tho.   my machine is Ubuntu 18.04.2 LTS, btw, if that matters.

davidbuzz, 2019-07-30:
I have fixed the missing Release file in ppa by editing these files:
sudo vi /etc/apt/sources.list.d/kevin-demarco-ubuntu-scrimmage-bionic*
and adding "[trusted=true]" after the 'ppa' and before the url.  I found the fix here: https://stackoverflow.com/questions/53800051/repository-does-not-have-a-release-file-error

davidbuzz, 2019-07-30:
after going back and stepping more carefully through the instructions again, I got error message saying that /usr/bin/vtk does not exist ( which is presumably to do with the tcl-vtk package not existing here ), but since I have the tcl-vtk6 package, I tried this hack: 
sudo ln -s /usr/bin/vtk6 /usr/bin/vtk

.. and now my 'cmake ..' command gives a scary message but completes:
The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.
...
-- Build files have been written to: /home/buzz/scrimmage/scrimmage/build

davidbuzz, 2019-07-30:
just a note here that after strugging through the above incorrect dependancies and symlink hacks, the 'make' command completed without error, and the command 'scrimmage ./missions/straight.xml' seems to startup the GUI ok, that's as far as my testing got.

SyllogismRXS, 2019-07-30:
If you are using bionic, you should add the scrimmage-bionic PPA (see: https://github.com/gtri/scrimmage#install-custom-built-binary-dependencies). You shouldn't have to create symlinks.

sudo add-apt-repository ppa:kevin-demarco/scrimmage-bionic

The tcl-vtk warning is fine and won't affect scrimmage. That's a warning that the package from Ubuntu isn't setup incorrectly.

closed 2020-09-30 by shaun-d-anderson