gtri/scrimmage #349: error occurs when trying to cmake a new project

LMozart, 2019-08-07:
I was trying to follow the tutorial to create a new plugin project. After generating a project by using script in the scripts folder, I started to cmake, but errors occurred. 
`
CMake Error at CMakeLists.txt:79 (include):
  include could not find load file:

    GenerateSetEnv

CMake Error at CMakeLists.txt:80 (GenerateSetEnv):
  Unknown CMake command "GenerateSetEnv".
`
Any suggestion?

SyllogismRXS, 2019-08-07:
Are you using Ubuntu? Were you able to build the core scrimmage project? 

GenerateSetEnv is a CMake macro provided by the core scrimmage project (https://github.com/gtri/scrimmage/blob/master/share/cmake/scrimmage/GenerateSetEnv.cmake). When you run CMake in the core scrimmage project, it will make the GenerateSetEnv available to other CMake projects that first all find_package(scrimmage).

SyllogismRXS, 2019-08-07:
By default, when you run "cmake" in the scrimmage, the cmake project will create a folder at "~/.cmake/packages/scrimmage", which contains a file that points at your scrimmage build directory. This is how other scrimmage/cmake projects find the original scrimmage project.

LMozart, 2019-08-08:
Thanks for answering. I am able to build the core scrimmage and I am using Ubuntu. I followed the tutorial below: `$ cd /path/to/scrimmage/scripts
$ mkdir ~/scrimmage
$ ./create-scrimmage-project.py my-scrimmage-plugins ~/scrimmage`
[http://www.scrimmagesim.org/sphinx/html/tutorials/create-project.html](url)
So here is the problem. I think the cmake process in my new project has already found the scrimmage core code. But still can not use GenerateSetEnv.