gtri/scrimmage #362: Execute "scrimmage ./missions/straight.xml" without GUI

BreederBai, 2019-09-17:
ubuntu 16.04
[image]
What should I do to solve this problem?

SyllogismRXS, 2019-09-17:
You need to install libvtk6-dev (Ubuntu Xenial) in order for the scrimmage viewer to be built. You can directly install VTK with the following command: sudo apt-get install libvtk6-dev. Then you need to run "cmake .." and "make in the scrimmage build directory again for cmake to find the newly installed VTK package. 

Based on the output of the install-binaries script, you may have some held debian packages. If you want to resolve that issue, you can manually try to install each of the packages listed (i.e., just try sudo apt-get install <package-name>, until you find the held package.) This may not be necessary depending on the plugins that you want to use.

BreederBai, 2019-09-23:
@SyllogismRXS Thank you

closed 2019-09-23 by SyllogismRXS