gtri/scrimmage #385: CMake: undefined references to Python libraries when linking scrimmage executable

mjdebord, 2019-11-04:
Answer the following questions:

* what are you trying to do?
Build Scrimmage with python 3.7 virtual environment. Ubuntu 18.04

* what is the problem and how can it be recreated?
Undefined reference errors to python 

I am using pyenv-virtualenv to manage my python virtual environments. I am attempting to use python 3.7.4

I used the following CMake command:

> cmake -DPYTHON_EXECUTABLE:FILEPATH=/home/[user]/.pyenv/versions/marl/bin/python -DPYTHON_INCLUDE_DIR:PATH=/home/[user]/.pyenv/versions/3.7.4/include/python3.7m -DPYTHON_LIBRARY:FILEPATH=/home/[user]/.pyenv/versions/3.7.4/lib/libpython3.so ..

CMake completes without errors or warnings (aside from VTK ubuntu bug)

Using make results in the following log:

> [  1%] Generating scrimmage/proto/Color.pb.cc, scrimmage/proto/Color.grpc.pb.cc, scrimmage/proto/Color.pb.h, scrimmage/proto/Color.grpc.pb.h
> [Generating scrimmmage/proto ...]
> [  4%] Generating scrimmage/proto/Visual.pb.cc, scrimmage/proto/Visual.grpc.pb.cc, scrimmage/proto/Visual.pb.h, scrimmage/proto/Visual.grpc.pb.h
> [  4%] Built target scrimmage-proto-gen
> Scanning dependencies of target scrimmage-protos
> [  4%] Building CXX object src/proto/CMakeFiles/scrimmage-protos.dir/scrimmage/proto/Color.pb.cc.o
> [Building Protos...]
> [ 10%] Building CXX object src/proto/CMakeFiles/scrimmage-protos.dir/scrimmage/proto/Visual.grpc.pb.cc.o
> [ 10%] Linking CXX shared library ../../lib/libscrimmage-protos.so
> [ 13%] Built target scrimmage-protos
> [ 14%] Generating scrimmage/msgs/Waypoint.pb.cc, scrimmage/msgs/Waypoint.grpc.pb.cc, scrimmage/msgs/Waypoint.pb.h, scrimmage/msgs/Waypoint.grpc.pb.h
> [Generating scrimmage/msgs]
> [ 17%] Generating scrimmage/msgs/Terrain.pb.cc, scrimmage/msgs/Terrain.grpc.pb.cc, scrimmage/msgs/Terrain.pb.h, scrimmage/msgs/Terrain.grpc.pb.h
> Scanning dependencies of target scrimmage-msgs
> [ 17%] Building CXX object msgs/CMakeFiles/scrimmage-msgs.dir/scrimmage/msgs/AuctionMsgs.pb.cc.o
> [Building Messages]
> [ 23%] Building CXX object msgs/CMakeFiles/scrimmage-msgs.dir/scrimmage/msgs/Waypoint.grpc.pb.cc.o
> [ 23%] Linking CXX shared library ../lib/libscrimmage-msgs.so
> [ 23%] Built target scrimmage-msgs
> Scanning dependencies of target scrimmage-core
> [ 24%] Building CXX object src/CMakeFiles/scrimmage-core.dir/autonomy/Autonomy.cpp.o
> [Build stuff]
> [ 36%] Building CXX object src/CMakeFiles/scrimmage-core.dir/common/WaypointListProcessor.cpp.o
> [ 36%] Linking CXX shared library ../lib/libscrimmage-core.so
> [ 36%] Built target scrimmage-core
> Scanning dependencies of target scrimmage-viewer
> [ 37%] Building CXX object src/viewer/CMakeFiles/scrimmage-viewer.dir/CameraInterface.cpp.o
> [ 37%] Building CXX object src/viewer/CMakeFiles/scrimmage-viewer.dir/Grid.cpp.o
> [ 37%] Building CXX object src/viewer/CMakeFiles/scrimmage-viewer.dir/OriginAxes.cpp.o
> [ 37%] Building CXX object src/viewer/CMakeFiles/scrimmage-viewer.dir/Updater.cpp.o
> [ 38%] Building CXX object src/viewer/CMakeFiles/scrimmage-viewer.dir/Viewer.cpp.o
> [ 38%] Linking CXX shared library ../../lib/libscrimmage-viewer.so
> [ 38%] Built target scrimmage-viewer
> Scanning dependencies of target scrimmage-bin
> [ 38%] Building CXX object share/scrimmage/CMakeFiles/scrimmage-bin.dir/main.cpp.o
**> [ 38%] Linking CXX executable ../../bin/scrimmage
> ../../lib/libscrimmage-core.so.0.0.1: undefined reference to `PyThread_get_key_value'
> ../../lib/libscrimmage-core.so.0.0.1: undefined reference to `PyUnicode_AsUTF8String'
> ../../lib/libscrimmage-core.so.0.0.1: undefined reference to `PyEval_GetBuiltins'
> ../../lib/libscrimmage-core.so.0.0.1: undefined reference to `PyExc_ValueError'
> [A lot more of these]**
> collect2: error: ld returned 1 exit status
> share/scrimmage/CMakeFiles/scrimmage-bin.dir/build.make:163: recipe for target 'bin/scrimmage' failed
> make[2]: *** [bin/scrimmage] Error 1
> CMakeFiles/Makefile2:112: recipe for target 'share/scrimmage/CMakeFiles/scrimmage-bin.dir/all' failed
> make[1]: *** [share/scrimmage/CMakeFiles/scrimmage-bin.dir/all] Error 2
> Makefile:151: recipe for target 'all' failed
> make: *** [all] Error 2

* what scrimmage commit are you on? You can see this with `git rev-parse HEAD`
commit 26bebb70fd8e5805f558db22a275bffc41946811, october 11th

* describe any changes you made to scrimmage.
For scrimmage/python in setup.py
```
    install_requires=[
        "numpy==1.16.4",
        "pandas==0.17.1",
        "protobuf==3.3.0",
        "argcomplete==1.10.0",
    ],
```
The versions were changed to >=, this is because tensorflow 2 (which I am hoping to utilize with scrimmage), requires newer versions of these libraries 
```
    install_requires=[
        "numpy>=1.16.4",
        "pandas>=0.17.1",
        "protobuf>=3.3.0",
        "argcomplete>=1.10.0",
    ],
```
No other changes were made

* If you can recreate the issue using only plugins in the scrimmage repository, what mission file are you running?

Thank you for any help!

SyllogismRXS, 2019-11-05:
Instead of manually specifying the python libraries and headers you can try to let CMake find the appropriate versions for you. I typically use "PYTHON_MIN_VERSION" CMake variable in the scrimmage project to specify the python version and it typically works for me.  For example...

$ cd /path/to/scrimmage/build
# source the virtual environment (I'm using virtualenvwrapper
$ workon tmp
# specify the python min version
$ cmake .. -DPYTHON_MIN_VERSION=3.6.8

You can then check that CMake found the appropriate headers, libraries, and executable:
-- Found PythonInterp: /home/myuser/python-envs/tmp/bin/python3.6 (found suitable version "3.6.8", minimum required is "3.6.8") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable version "3.6.8", minimum required is "3.6.8") 
-- Python Versions Found: 3.6.8
-- PYTHON_LIBRARIES : /usr/lib/x86_64-linux-gnu/libpython3.6m.so
-- PYTHON_INCLUDE_DIRS: /usr/include/python3.6m

We use virtualenvwrapper (https://virtualenvwrapper.readthedocs.io/en/latest/), scrimmage, and tensorflow together

mjdebord, 2019-11-05:
Thank you for your response, unfortunately just setting PYTHON_MIN_VERSION did not solve my problem (CMake could find the python executable by itself, but not the library), but it did prompt me to find a solution. It seems that the recent versions of pyenv-virtual may cause problems with CMake find_package(PythonLibs), to overcome this I used the following cmake command after activating my python virtual env (pyenv activate marl):

> cmake \
>   -DPYTHON_MIN_VERSION=3.7.4 \
>   -DPYTHON_LIBRARY=$(pyenv virtualenv-prefix)/lib/libpython3.7m.so \
>   -DPYTHON_INCLUDE_DIR=$(pyenv virtualenv-prefix)/include/python3.7m \
>   ..

This seems to have solved the problem and I now have scrimmage working (although I did have a second error about missing JSBsim, I had to sudo apt-get install scrimmage-jsbsim in order to fix that). I'm having a second issue with "Multiple Runs in Parallel" tutorial but I will open a second issue for that.

closed 2019-11-05 by mjdebord