gtri/scrimmage #390: scrimmage build fails on JSBSim API calls

dmagree, 2019-11-08:
On Ubuntu 18.04. Occurs since JSBSim upgrade commit at 82d839a6. 

Have tried sudo apt install scrimmage-jsbsim, get the following output.
```
scrimmage-jsbsim is already the newest version (0.1.5-5ppa0).
```
Have also removed and then installed scrimmage-jsbsim

SyllogismRXS, 2019-11-08:
The JSBSim version was upgraded to the newest version (which changed the API slightly) and based on a different request, I put all packages for xenial and bionic under the same Launchpad PPA: https://launchpad.net/~kevin-demarco/+archive/ubuntu/scrimmage, so the separate "scrimmage-bionic" PPA will be deprecated. I plan on leaving the scrimmage-bionic PPA for a few more months.

To fix this on your system: remove the old scrimmage-bionic reference (e.g., rm /etc/apt/sources.list.d/kevin-demarco-ubuntu-scrimmage-*.list) and just add the scrimmage ppa, which will cover both xenial and bionic (e.g., sudo add-apt-repository ppa:kevin-demarco/scrimmage).

The newest JSBSim package should be "0.2.1-1-bionicppa0 " on bionic.

Let me know if you have any problems.

dmagree, 2019-11-08:
That fixed it, thanks!

closed 2019-11-08 by dmagree