gtri/scrimmage #419: Can agents/entities be spawned at runtime?

TheDash, 2020-03-04:
Currently it looks like SCRIMMAGE must take an .xml config file to get it going - is it possible to spawn agents in at a later date and modify the simulation at runtime?

SyllogismRXS, 2020-03-05:
Yes, entities can be generated later at run-time if their entity description is provided in the mission xml file. Any plugin can publish on the "generate_entities" topic, which the main simulation loop receives and generates an entity. Take a look at the Straight.cpp example:

https://github.com/gtri/scrimmage/blob/master/src/plugins/autonomy/Straight/Straight.cpp#L198
https://github.com/gtri/scrimmage/blob/master/src/plugins/autonomy/Straight/Straight.cpp#L239

You also have to specify the entity description in the scrimmage mission xml file, with a <count> of 0, so the entity is not generated at the start of the simulation:
https://github.com/gtri/scrimmage/blob/master/src/plugins/autonomy/Straight/Straight.cpp#L239

The entity tag is used to specify which entity to generate in the "generate_entities" message:
<entity tag="gen_straight">

TheDash, 2020-03-07:
I'm kind of new to this - if an entity is described in the mission XML file, does that mean I can spawn X number of compies of it later using the above method? Or just spawn that one one time at a later date?

SyllogismRXS, 2020-03-09:
Every time you publish to the "generate_entities" topic, you will be able to spawn an additional entity. So, if within a single time step, your plugin publishes multiple messages to the "generate_entities" topic, multiple entities will be generated.

TheDash, 2020-03-15:
What should I be using/linking against to write code that will publish against the generate_entities topic? Are there CLI tools to debug this/see this for a running SCRIMMAGE simulator?

I couldn't find anything in the documentation marking this functionality, any help would be appreciated!

TheDash, 2020-03-25:
I tried with <count>0</count> but that gives me a segfault. Am I missing something? I followed the tutorial about this but when count is 0 I segfault

TheDash, 2020-03-25:
#434 

I investigated this further and made a bug report.

SyllogismRXS, 2020-03-25:
Closing this since it is being tracked in another issue.

closed 2020-03-25 by SyllogismRXS