gtri/scrimmage #432: id_to_ent_map_ isn't getting set for autonomy plugins

jharrison, 2020-03-24:
Entity::init initializes the `id_to_ent_map_` for sensor, motion_model, and controller plugins:

```c++
            sensor->set_id_to_ent_map(id_to_ent_map);
```

```c++
            motion_model_->set_id_to_ent_map(id_to_ent_map);
```

```c++
            controller->set_id_to_ent_map(id_to_ent_map);
```

But it does not do the same for autonomy plugins (nor is it done in make_autonomy). I was able to fix the problem by adding the following change at [line 424](https://github.com/gtri/scrimmage/blob/master/src/entity/Entity.cpp#L424):

```c++
        if (autonomy) {
            (*autonomy)->set_id_to_ent_map(id_to_ent_map);   // ADD THIS LINE
            autonomies_.push_back(*autonomy);
        }
```

According to @SyllogismRXS:

> This looks like a bug that creeped in when the Entity.cpp class was refactored. The id_to_ent_map should be added as an argument to the make_autonomy function that is defined in EntityPluginHelper.h. In the make_autonomy function, the id_to_ent_map should be set.

SyllogismRXS, 2020-03-25:
This should be fixed in pull request: https://github.com/gtri/scrimmage/pull/433
Try it out and let me know if this fixes your problem.

jharrison, 2020-04-02:
Looks to me like it's working.

closed 2025-03-03 by wfsyre