gtri/scrimmage #434: Spawning entities at runtime bug

TheDash, 2020-03-25:
Answer the following questions:

* what are you trying to do?
Spawn an agent at runtime

* what is the problem and how can it be recreated?
When <count> is 0 in the XML file, it sets max_num_entities in SimControl.cpp to be 0. Therefore, it is impossible to spawn agents without causing a segfault.

* what scrimmage commit are you on? You can see this with `git rev-parse HEAD`
Head

* describe any changes you made to scrimmage.
I changed max_num_entities to 100 for example, and it continued on without segfaulting

Changes proposed:
Allow there to be a parameter in the XML file such as "max_num_entities" which is separate from count and outside of the entity tag such that spawning entities at runtime is possible

SyllogismRXS, 2020-03-25:
I made some changes to how the scrimmage RTree is constructed in SimControl. I merged the changes into master after they passed Travis CI tests. See the changes in commit https://github.com/gtri/scrimmage/pull/435/commits/bdebe85d8b7054b1376b441950866b43bb78e47f

Let me know if this works for you.

TheDash, 2020-03-25:
Thank you!

I will get back to you tomorrow on the changes.

wfsyre, 2025-03-03:
Assuming this was fixed since we haven't hear anything

closed 2025-03-03 by wfsyre