gtri/scrimmage #440: env default python should be standard, not 2.7

tm132, 2020-04-03:
Scrimmage's default cmake configuration should use the default version of python setup in the user's environment (hopefully 3.x), instead of explicitly using python 2.7, since 2.7 is deprecated.

i.e., if you run...
`$ cmake ..`
...it should build with whatever version of python they have selected as the default in their environment.

For instance, mine is set to:
```
$ python -V
Python 3.7.3
$ which python
/home/user/anaconda3/bin/python
$
```

-+-

The cmake files should be updated, and then the README.md should be updated, something like this, if you wish to allow users to target other/older versions of python:

> You can specify a target Python version in SCRIMMAGE core by setting the PYTHON_TARGET_VERSION cmake variable. For example, to specify a Python version of 2.7, first clear the cmake cache file and rerun cmake:
```
$ rm CMakeCache.txt
$ cmake .. -DPYTHON_TARGET_VERSION=2.7
```

mday299, 2020-09-29:
@tm132 finally getting to this.  Is this still an issue?

tm132, 2020-10-15:
No one contacted me saying they were fixing it, so it probably still is. Glancing over the CMakeLists.txt file, it looks like it might still be a problem. I might have time to test this in the next couple of days to re-verify.

tm132, 2021-03-26:
An update: this problem still exists:

```
scrimmage@orca:~/sandbox/scrimmage/build$ cmake ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
// HERE -----------------------------------------------
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.17", minimum required is "2.7")
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.17", minimum required is "2.7")
-- Python Versions Found: 2.7.17
-- PYTHON_LIBRARIES : /usr/lib/x86_64-linux-gnu/libpython2.7.so
-- PYTHON_INCLUDE_DIRS: /usr/include/python2.7
// -----------------------------------------------------
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Protobuf: /opt/scrimmage/x86_64-linux-gnu/lib/libprotobuf.so;-lpthread (found suitable version "3.3.0", minimum required is "3.0") 
-- Found GeographicLib: /usr/include (found version "1.49") 
-- Found JSBSIM 
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- Found VTK Version: 6.3.0
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   date_time
--   iostreams
--   program_options
--   regex
--   filesystem
--   system
--   graph
--   chrono
--   atomic
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- LTO enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /home/scrimmage/sandbox/scrimmage/build
scrimmage@orca:~/sandbox/scrimmage/build$ 
```
My environment settings:
```
scrimmage@orca:~/sandbox/scrimmage/build$ python -V
Python 3.7.3
scrimmage@orca:~/sandbox/scrimmage/build$ python3 -V
Python 3.6.9
scrimmage@orca:~/sandbox/scrimmage/build$ 
```

PR #521 (2021-12-23): CMake updates to use environment Python by default
Closes: #440 
Changes:
- Primarily, the main `CMakeLists.txt` was updated to remove the setting of `PYTHON_MIN_VERSION` to 2.7 by default.
- The `README.md` was updated to clarify what the `PYTHON_MIN_VERSION` flag is used for.

Functionality changes:
- `PYTHON_MIN_VERSION` was repurposed to not be set to anything by default, and instead allows the user to optionally specify, if they would prefer, a different python version to use than the environment default. For example, the latest anaconda (2020.11) comes with python 3.8.
- By default (no v [... cut]
closed 2021-12-23 by shaun-d-anderson