gtri/scrimmage #445: Bin files being created despite output_type of nothing

laurathepluralized, 2020-05-04:
Answer the following questions:

* What are you trying to do?
Run SCRIMMAGE without generating `.bin` output files.

* What is the problem and how can it be recreated?
Recreation steps, verified on two systems, both running Ubuntu 18.04:
1. Modify `missions/straight-no-gui.xml` to have only one `output_type`, either `nothing` or `summary`, as below:
```
<output_type>nothing</output_type>
# OR
<output_type>summary</output_type>
```
2. Run 
```
$ source ~/.scrimmage/setup.bash && scrimmage missions/straight-no-gui.xml
```
and verify that the command line output looks correct for running this mission; it looked fine for me.

3. Check that run's results directory (while I did use `~/.scrimmage/logs/latest`, I made sure to verify that the timestamp of the results directory was indeed for the run in question)

4. Results directory is filled with the following:
```
contact_visual.bin
frames.bin
log.txt
mission.orig.xml
mission.xml
runtime_seconds.txt
shapes.bin
summary.csv  # as expected, this file was only present when output_type was summary
utm_terrain.bin
```
Unfortunately, `.bin` files are still being generated even when I choose an `output_type` that shouldn't let them be saved. This problem even persists if I delete the `<terrain>` line in the mission file (my sanity check to ensure SCRIMMAGE has absolutely no reason to save any terrain-related `.bin` files).

* What scrimmage commit are you on? You can see this with `git rev-parse HEAD`
f9cd355ab36d87a76c15f51b5dfc55a1b94026cb

* Describe any changes you made to scrimmage.
  * Modifications to `missions/straight-no-gui.xml` (see above)
  * SCRIMMAGE was compiled with gRPC enabled and with Python bindings for Python3.6

* If you can recreate the issue using only plugins in the scrimmage repository, what mission file are you running?
`straight-no-gui.xml`, modified as above

I may take a look at MissionParse and other code that interacts with it to see if I can figure out why these `.bin` files are still getting created, though if you get a chance to take a look into this, @SyllogismRXS, I would appreciate it--you probably know much more readily where to start looking to fix this! :) Please let me know if I can provide any more info for debugging. Thank you!

laurathepluralized, 2020-05-04:
Oops, I just noticed that for outputting nothing, the `output_type` line should be
```
<output_type></output_type>
```
The above line means that no log directory is created at all. I'll edit my earlier comment (and issue title, if I can do that) to reflect this. I'd still like to figure out why `.bin` files are being created when I'm not asking SCRIMMAGE to output `frames` or `all`, though. Thanks!

mday299, 2020-09-29:
@laurathepluralized cleaning up issues in scrimmage.  Is this still one?

laurathepluralized, 2020-09-29:
Hi @mday299, thanks for following up. Yes, as of the most recent commit on master (3f69fb430), I'm still seeing this issue, and haven't had the bandwidth to dig around for the cause of it myself. Thanks!

mday299, 2020-12-10:
I have not had time to look at this. If this is still a priority, please let me know.

wfsyre, 2025-03-03:
Please use <no_bin_logging>true</no_bin_logging> to prevent scrimmage from generating bin files

closed 2025-03-03 by wfsyre