gtri/scrimmage #451: GUI won't pop up

johndlok, 2020-05-18:
The program won't launch the GUI for me, hangs up when I run these lines:
$ cd </path/to/>scrimmage
$ scrimmage ./missions/straight.xml

I'm running Bionic Beaver.
 
I've ran these steps:  
    1  mkdir -p ~/scrimmage && cd ~/scrimmage
    2  git clone https://github.com/gtri/scrimmage.git
    3  sudo apt install git
    4  git clone https://github.com/gtri/scrimmage.git
    5  cd scrimmage
    6  sudo ./setup/install-binaries.sh -e 0 -p 3
    7  sudo add-apt-repository ppa:kevin-demarco/scrimmage
    8  sudo apt-get update
    9  sudo apt-get install scrimmage-dependencies scrimmage-jsbsim
   10  source /opt/scrimmage/*/setup.sh
   11  mkdir build && cd build
   12  cmake ..
   13  make
   14  source ~/.scrimmage/setup.bash
   15  echo "source ~/.scrimmage/setup.bash" >> ~/.bashrc
   16  cd scrimmage

I'm stuck on step "$ cd </path/to/>scrimmage" in the section: "Run SCRIMMAGE".
I get the error bash: "bash: scrimmage: Is a directory.

What I tried so far: run "$ cd scrimmage" twice, then
"~/scrimmage/scrimmage$ scrimmage ./missions/straight.xml" which runs but it doesn't open the GUI, just hangs up and appears frozen.

What do you recommend checking to finish installing?

Scrimmage commit from `git rev-parse HEAD`:
264003a786f05354d5c576c103df368966aebeb4

SyllogismRXS, 2020-05-18:
Can you run the straight_jsbsim.xml mission? For example: scrimmage ./missions/straight_jsbsim.xml

Are you sure everything built successfully? (e.g., after you ran the "make" command, there weren't any errors).

What is your Linux command line experience level?

johndlok, 2020-05-18:
Just starting out, and I'm willing to learn whatever is needed.

I cannot run that mission. I'm not positive there were no errors. I'll rename the original folder and try again from step 1.

closed 2020-05-19 by johndlok