gtri/scrimmage #474: doesn't run setup.sh.in and

sbyebss, 2020-07-25:
* what are you trying to do?

  I am trying to [Setup SCRIMMAGE/AirSim on my Linux machine.](http://www.scrimmagesim.org/sphinx/html/tutorials/airsim-plugin.html#setup-scrimmage-airsim-on-your-linux-machine) 
 
* what is the problem and how can it be recreated?

    I followed all commands and finally run `source ~/.scrimmage/setup.bash` but in terminal, it shows `~/.scrimmage folder doesn't exist`.  I think the problem is it didn't run [setup.sh.in](https://github.com/gtri/scrimmage/blob/0861cdc21d39c9a662fc818a4d73f48081b64017/3rd-party/packages/scrimmage-dependencies/scrimmage-dependencies/scripts/setup.sh.in).

* what scrimmage commit are you on? You can see this with `git rev-parse HEAD`

2142936b6722b7e20b77cc71975d102106277958

* describe any changes you made to scrimmage.

  I manually create `${HOME}/.scrimmage/setup.bash` and echo the code into this file.

SyllogismRXS, 2020-07-26:
Those instructions might be slight out of date or they assumed you had already built scrimmage at some point. You need to run the "cmake .." command in the build directory before the ~/.scrimmage/setup.bash file is generated. I recommend building scrimmage based on the main README's instructions first (https://github.com/gtri/scrimmage#build-scrimmage), then to build the AirSim plugin, you can look at the additional steps in the tutorials.

sbyebss, 2020-07-27:
okay, thanks! The README and the document are different. Get it.

closed 2020-07-27 by sbyebss