gtri/scrimmage #475: Request an option to format CSV output.

goehle, 2020-07-29:
* what are you trying to do?  We are printing debugging output to a CSV using the common::CSV class.  The debugging information includes reported latitude and longitude at each time step. 

* what is the problem and how can it be recreated?  CSV uses std::to_string to print the rows which produces 6 decimal places.   This isn't enough for latitudes and longitudes.  For debugging purposes we usually use 12 decimal places.  One solution would be to modify CSV to use something like sprintf and allow for formatting fields. Another solution would be to modify CSV to support string inputs to rows so that entries could be preformatted.  

* what scrimmage commit are you on? You can see this with `git rev-parse HEAD`
cb566bcd9b6a33a98b607dd3239b9c09bc814382

* describe any changes you made to scrimmage.
None.  We could add the feature ourselves but don't want to fork the CSV library.  

* If you can recreate the issue using only plugins in the scrimmage repository, what mission file are you running?
N/A

mday299, 2020-09-29:
@goehle Confirming: is this the Apache library at https://commons.apache.org/proper/commons-csv/

If so, what is preventing you from submitting a patch yourself to https://github.com/apache/commons-csv ?

goehle, 2020-09-29:
I was referring to https://github.com/gtri/scrimmage/blob/master/src/common/CSV.cpp  Now that you suggest it though, I should  amend my feature request to just replace the homegrown CSV library with a wrapper around a more fully featured one.