gtri/scrimmage #481: test_opneai is failed at "Build and Run Tests "

kkz003, 2020-09-18:
Hi! 
I am installing "scrimmage" following the procedure on https://github.com/gtri/scrimmage.
I use Ubuntu 18.04 via virtual box on mac. 
At the "Run SCRIMMAGE" the simulation is displayed successfully.
However at "Build and Run Tests " step,  "make test" includes 1 failed test,
 test_openai.  

I would like to make test_openai. Could you give me any suggestions and instructions. I have no knowledge of "make" files but I can edit them.

Sincerely.

---------------------------------------------------------------
VirtualBoxUB18:~/scrimmage/scrimmage/build2$ make test
Running tests...
Test project /home/masaok/scrimmage/scrimmage/build2
      Start  1: test_csv
 1/18 Test  #1: test_csv .........................   Passed    0.05 sec
      Start  2: test_algorithms
 2/18 Test  #2: test_algorithms ..................   Passed    0.01 sec
      Start  3: test_angles
 3/18 Test  #3: test_angles ......................   Passed    0.15 sec
      Start  4: test_collisions
 4/18 Test  #4: test_collisions ..................   Passed    1.49 sec
      Start  5: test_delayed_task
 5/18 Test  #5: test_delayed_task ................   Passed    0.03 sec
      Start  6: test_exponential_filter
 6/18 Test  #6: test_exponential_filter ..........   Passed    0.05 sec
      Start  7: test_find_mission
 7/18 Test  #7: test_find_mission ................   Passed    0.02 sec
      Start  8: test_id
 8/18 Test  #8: test_id ..........................   Passed    0.03 sec
      Start  9: test_params
 9/18 Test  #9: test_params ......................   Passed    0.03 sec
      Start 10: test_quaternion
10/18 Test #10: test_quaternion ..................   Passed    0.02 sec
      Start 11: test_rtree
11/18 Test #11: test_rtree .......................   Passed    0.10 sec
      Start 12: test_simple
12/18 Test #12: test_simple ......................   Passed    0.01 sec
      Start 13: test_state
13/18 Test #13: test_state .......................   Passed    0.04 sec
      Start 14: test_utilities
14/18 Test #14: test_utilities ...................   Passed    0.03 sec
      Start 15: test_entity_configs
15/18 Test #15: test_entity_configs ..............   Passed    1.21 sec
      Start 16: test_openai
16/18 Test #16: test_openai ......................***Exception: Child aborted  0.95 sec
      Start 17: test_shape
17/18 Test #17: test_shape .......................   Passed    0.02 sec
      Start 18: test_parse_utils
18/18 Test #18: test_parse_utils .................   Passed    0.03 sec

94% tests passed, 1 tests failed out of 18

Total Test time (real) =   4.30 sec

The following tests FAILED:
     16 - test_openai (Child aborted)
Errors while running CTest
Makefile:129: recipe for target 'test' failed
make: *** [test] Error 8

frazierbaker, 2020-09-29:
Hi @kkz003!  Thanks for using SCRIMMAGE.

Can you confirm that you kept the `-DPYTHON_MIN_VERSION=3.6` flag on `cmake` when building your tests?  My first guess looking at your test output is that python bindings may not have been turned on when building the tests.  OpenAI plugins require python bindings in order to work.

Try re-building your tests with 
```
cmake .. -DPYTHON_MIN_VERSION=3.6 -DBUILD_TESTS
make test
```

If that works, we can discuss updating the README to reflect the appropriate CMake flags.

frazierbaker, 2021-02-22:
Update: With python3 you either need to make sure that `python` points to python3 in your path (the same one you have gym installed on) or you need to also `pip install gym` on python version 2 in order for this to work.

frazierbaker, 2021-05-03:
This fixed the error for me.  If you continue to have trouble after installing with python and ensuring the openai gym is installed on the right version of python, feel free to write back on this thread and we can re-open the issue.

closed 2021-05-03 by frazierbaker