gtri/scrimmage #482: Failed to open autonomy plugin: ROSAirSim

chao8038, 2020-09-24:
Answer the following questions:

* what are you trying to do?
Using ROSAirSim plugin to publish Airsim/Unreal data to ROS.
* what is the problem and how can it be recreated?
Flowing the https://www.scrimmagesim.org/sphinx/html/tutorials/airsim-plugin.html step to use ROSAirSim plugin, but I saw the problem about it. I can find the ROSAirSim_plugin folder in my project, so I have no idea to solve it.
* what scrimmage commit are you on? You can see this with `git rev-parse HEAD`
23cd136ae655297ee455d561c63a62daad134f7f.
* describe any changes you made to scrimmage.
build and run with ROS.
* If you can recreate the issue using only plugins in the scrimmage repository, what mission file are you running?
I run the quad-airsim-ex1.xml mission and show the message about the following.
[AirSimSensor] Vehicle Name: robot1
[AirSimSensor] Saving camera images and airsim_data.csv of pose to SCRIMMAGE Logs Directory.
[AirSimSensor] Retrieving image data within AirSimSensor::request_images() thread.
[AirSimSensor] Image Acquisition Period = 0.33
[AirSimSensor] Adding camera to Vehicle 'robot1' Camera_Name=front_center, Image_Type_Num=0, Image_Type_Name=Scene, Height=144, Width=256, FOV=90
[AirSimSensor] Adding camera to Vehicle 'robot1' Camera_Name=front_center, Image_Type_Num=2, Image_Type_Name=DepthPerspective, Height=144, Width=256, FOV=90
[AirSimSensor] Adding camera to Vehicle 'robot1' Camera_Name=front_center, Image_Type_Num=5, Image_Type_Name=Segmentation, Height=144, Width=256, FOV=90
[AirSimSensor] Adding camera to Vehicle 'robot1' Camera_Name=front_center, Image_Type_Num=7, Image_Type_Name=Infrared, Height=144, Width=256, FOV=90
[AirSimSensor] Retrieving LIDAR data within AirSimSensor::request_lidar() thread.
[AirSimSensor] LIDAR Acquisition Period = 0.1
[AirSimSensor] Adding LIDAR sensor 'lidar1' to Vehicle 'robot1'.
[AirSimSensor] Retrieving IMU data within AirSimSensor::request_imu() thread.
[AirSimSensor] IMU Acquisition Period = 0.1
[AirSimSensor] Adding IMU sensor 'imu1' to Vehicle 'robot1'.
Failed to open autonomy plugin: ROSAirSim
robot1 Sim Client waiting for Unreal/AirSim connection - 
X[AirSimSensor] Sim Client for robot1 connected to AirSim: ip 127.0.0.1, port 41451

nrakoski3, 2020-10-02:
Please make sure you rebuilt SCRIMMAGE with ROS_PLUGINS turned on and that you have a ROSCORE running in a different terminal:

```
# Go into Scrimmage
cd /path/to/scrimmage/
# Delete build directory
rm -rf build/ && mkdir build/ && cd build/

# Config CMake to build SCRIMMAGE ROS Plugins
cmake .. -DAIRSIM_ROOT_SEARCH=/home/nrakoski3/scrimmage/AirSim/ -DROS_VERSION=melodic -DBUILD_ROS_PLUGINS=ON
# Build
make -j7

# Open a second Terminal window and start ROS
roscore

# Run from original Terminal window
cd .. && scrimmage ./missions/quad-airsim-ex1.xml

# Press 'b' on your keyboard to unpause the simulation.
# More about SCRIMMAGE controls here: https://github.com/gtri/scrimmage#gui-commands
```

Also make sure that you have uncommented the ROSAirSim tag in the quad-airsim-ex1.xml XML file. Though if you got this error you should have it uncommented.

Let me know if this helps, best,

Natalie

chao8038, 2020-10-02:
Yes, I follow the steps and rebuild it many times. After that I tried, the result is the same as above.

closed 2020-10-06 by chao8038

nrakoski3, 2020-11-17:
Hello chao8038,

I am sorry for having trouble recreating your issue before. I was able to recreate and identify the bug. To fix for now use this commit of scrimmage master and it will work: ad18e6c2a7ea01de822325745d9a247ffad51d08

More details in this ticket: https://github.com/gtri/scrimmage/issues/498