gtri/scrimmage #483: Failed source update

Pravin854, 2020-09-30:
After adding the package to the repository using 'sudo add-apt-repository ppa:kevin-demarco/scrimmage'

I've tried updating the source with the command, `sudo apt-get update`

But it gave me the following error,

'Ign:11 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy InRelease                                                                      
Ign:12 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy Release                                                                        
Ign:13 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Sources                                                                   
Ign:14 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main amd64 Packages                                                            
Ign:15 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main i386 Packages                                                             
Ign:16 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main all Packages                                                              
Ign:17 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Translation-en_IN                          
Ign:18 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Translation-en
Ign:19 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main amd64 DEP-11 Metadata
Ign:20 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main DEP-11 64x64 Icons
Ign:13 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Sources
Ign:14 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main amd64 Packages                             
Ign:15 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main i386 Packages                              
Ign:16 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main all Packages
Ign:17 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Translation-en_IN                          
Ign:18 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Translation-en
Ign:19 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main amd64 DEP-11 Metadata
Ign:20 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main DEP-11 64x64 Icons
Ign:13 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Sources                                    
Ign:14 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main amd64 Packages                             
Ign:15 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main i386 Packages                              
Ign:16 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main all Packages                               
Ign:17 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Translation-en_IN
Ign:18 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Translation-en
Ign:19 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main amd64 DEP-11 Metadata
Ign:20 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main DEP-11 64x64 Icons
Ign:13 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Sources
Ign:14 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main amd64 Packages
Ign:15 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main i386 Packages
Ign:16 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main all Packages
Ign:17 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Translation-en_IN
Ign:18 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Translation-en                             
Ign:19 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main amd64 DEP-11 Metadata                      
Ign:20 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main DEP-11 64x64 Icons
Ign:13 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Sources
Ign:14 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main amd64 Packages
Ign:15 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main i386 Packages
Ign:16 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main all Packages
Ign:17 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Translation-en_IN
Ign:18 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Translation-en
Ign:19 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main amd64 DEP-11 Metadata
Ign:20 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main DEP-11 64x64 Icons
Err:13 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Sources
  404  Not Found [IP: 2001:67c:1560:8008::15 80]
Ign:14 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main amd64 Packages
Ign:15 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main i386 Packages
Ign:16 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main all Packages
Ign:17 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Translation-en_IN
Ign:18 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main Translation-en
Ign:19 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main amd64 DEP-11 Metadata
Ign:20 http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy/main DEP-11 64x64 Icons
Hit:21 http://in.archive.ubuntu.com/ubuntu xenial InRelease
Hit:22 http://in.archive.ubuntu.com/ubuntu xenial-updates InRelease
Hit:23 http://in.archive.ubuntu.com/ubuntu xenial-backports InRelease
Get:24 http://in.archive.ubuntu.com/ubuntu xenial/main Sources [868 kB]
Get:25 http://in.archive.ubuntu.com/ubuntu xenial/multiverse Sources [179 kB]
Get:26 http://in.archive.ubuntu.com/ubuntu xenial/restricted Sources [4,808 B]           
Get:27 http://in.archive.ubuntu.com/ubuntu xenial/universe Sources [7,728 kB]              
Get:28 http://in.archive.ubuntu.com/ubuntu xenial-updates/main Sources [528 kB]            
Get:29 http://in.archive.ubuntu.com/ubuntu xenial-updates/multiverse Sources [12.2 kB]  
Get:30 http://in.archive.ubuntu.com/ubuntu xenial-updates/restricted Sources [3,608 B]    
Get:31 http://in.archive.ubuntu.com/ubuntu xenial-updates/universe Sources [418 kB]       
Get:32 http://in.archive.ubuntu.com/ubuntu xenial-backports/main Sources [6,720 B]      
Get:33 http://in.archive.ubuntu.com/ubuntu xenial-backports/universe Sources [11.0 kB]
Fetched 109 kB in 2min 13s (819 B/s)
Reading package lists... Done
W: The repository 'http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu groovy Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://in.archive.ubuntu.com/ubuntu/dists/xenial/InRelease  Unable to find expected entry 'contrib/source/Sources' in Release file (Wrong sources.list entry or malformed file)
E: Failed to fetch http://in.archive.ubuntu.com/ubuntu/dists/xenial-updates/InRelease  Unable to find expected entry 'contrib/source/Sources' in Release file (Wrong sources.list entry or malformed file)
E: Failed to fetch http://in.archive.ubuntu.com/ubuntu/dists/xenial-backports/InRelease  Unable to find expected entry 'contrib/source/Sources' in Release file (Wrong sources.list entry or malformed file)
E: Failed to fetch http://security.ubuntu.com/ubuntu/dists/xenial-security/InRelease  Unable to find expected entry 'contrib/source/Sources' in Release file (Wrong sources.list entry or malformed file)
E: Failed to fetch http://ppa.launchpad.net/kevin-demarco/scrimmage/ubuntu/dists/groovy/main/source/Sources  404  Not Found [IP: 2001:67c:1560:8008::15 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.'

shaun-d-anderson, 2020-09-30:
Can you confirm what version of Ubuntu you are running for your installation of Scrimmage? Ubuntu 20.10 (Groovy Gorilla) is not supported at this time.

Pravin854, 2020-09-30:
I'm using ubuntu 16.04 (Xenial Xerus).

shaun-d-anderson, 2020-09-30:
Can you provide additional details to your problem is it repeatable?  Were there any errors when you ran: ``sudo ./setup/install-binaries.sh -e 0 -p 3``

I confirmed, that the current master commit, installs and functions on clean install of ubuntu 16.04 (Xenial Xerus) without issue posted above.

Pravin854, 2020-09-30:
`sudo ./setup/install-binaries.sh -e 0 -p 3` command is working fine. All packages have been installed. I'll try installing it on some other system and see if I get the same error. Thank you for the prompt response.

mday299, 2020-11-04:
Please confirm that this is no longer an issue.

closed 2020-11-05 by Pravin854