gtri/scrimmage #489: Terrain Lighting Glitch

stoplime, 2020-10-13:
Low priority
Lighting Bug:
The terrain can get very dark by rotating the camera to view the horizon, parallel to the ground.
To repeat, change the camera to free mode. Rotate and move the camera near the ground and point the camera to the horizon. The terrain on the ground will dim based on the camera's angle.

stoplime, 2020-10-13:
A potential solution could be baked lighting onto the terrain similar to how baked lighting works in Unity. Simply put, the lighting effects gets added to the texture of the terrain and so the engine does not need to dynamically calculate the lighting but still has decent results assuming there are no dynamic light sources or dynamic shadows.

mday299, 2020-11-04:
Again, this is a major undertaking. SCRIMMAGE ideally, would not be used for rendering anything and we would use something like OpenSceneGraph, Unity, or Unreal for rendering.