gtri/scrimmage #491: Saved Camera Orientations

stoplime, 2020-10-13:
Low priority
Camera utility:
Have a set of saved camera orientations that can be easily switched to with hotkeys. For example, have a top-down orientation where the camera is looking straight down over (0, 0) and North is up, then have a key ("0" for example) that can be pressed to teleport the camera to that position and orientation.

This would be low priority as a nice to have to easily have a reset for the camera to go back to.

frazierbaker, 2020-11-04:
I could see how camera presets could be useful, but presets' utility would depend on the mission and entity positioning.  I could see how a reset hotkey could be useful.  If you'd like to submit a PR, we can discuss merging this in for the SCRIMMAGE viewer.

wfsyre, 2025-03-03:
pressing "u" now undoes the last camera translation (not rotation).

closed 2025-03-03 by wfsyre