gtri/scrimmage #505: Radius Tag Definition missing from Mission XML Documentation

christomaszewski, 2020-12-30:
I'm running through all the documentation and tutorials and noticed this tag is present on the entities in the straight-vs-motorschemas mission but not defines in the Mission XML documentation found here: http://www.scrimmagesim.org/sphinx/html/overview/xml-tags.html

If anyone can provide me with a clear description of what this tag defines I can add it into the documentation updates I'm working on.

SyllogismRXS, 2021-01-01:
The radius tag can be used as a first-order "size" approximation of an entity. For example, the ContactBlobCamera sensor uses the radius of an entity to generate a bounding box around other entities: https://github.com/gtri/scrimmage/blob/master/src/plugins/sensor/ContactBlobCamera/ContactBlobCamera.cpp#L293 

The radius isn't used in the core simulation engine.

christomaszewski, 2021-01-02:
Ok cool I can add that to the mission tag xml doc. Also the agents in the straight-vs-motorschemas mission have different radii set but I don’t think the interaction plugins set don’t actually check each radius of each agent when checking for collisions. I was interested in why each group of agents on each team had different sizes but maybe it’s just a typo.

SyllogismRXS, 2021-01-04:
The collision detection doesn't use the entity radius, just the collision_range parameter: https://github.com/gtri/scrimmage/blob/master/src/plugins/interaction/SimpleCollision/SimpleCollision.cpp#L59

Including the "radius" of the entity in the XML was potentially a mistake, but we added it early in the development. It's only used by a few plugins. We probably should have come up with a way for plugins to expect entities to have additional required XML tags, but not require those tags to be processed by the core MissionParser code.

christomaszewski, 2021-01-04:
Ok, I was just asking as part of getting up to speed. I think there may be some use cases where the entities are of multiple different sizes where we may want to make the collision_range parameter be on top of the radius of each entity but I suppose we can make that its own interaction plugin. I'll add the documentation for the entity radius tag to my documentation updates. I'll close this issue after I put in a PR.

christomaszewski, 2021-01-05:
@SyllogismRXS the plot_tracks tag definition is also missing from the documentation. Do you know if this parameter is still used and what exactly it does when enabled?

SyllogismRXS, 2021-01-05:
The purpose of this parameter was to plot the x/y/z positions of the entities immediately following the simulation. It was used early in the project to verify motion models, but since then we have suggested that developers script this sort of behavior on their own by calling scrimmage from a script and using the scrimmage-python package to plot trajectories on their own. This parameter could probably be deprecated because I doubt anyone is using it.