gtri/scrimmage #530: Controller step function provides incorrect parameters with motion multiplier

goehle, 2021-08-02:
Answer the following questions:

* what are you trying to do? Run a controller with a motion multiplier and utilize the "t and dt" parameters provided by the step function

* what is the problem and how can it be recreated?  The step function for a controller provides the _unmodified_ time and timestep parameters, even when a `motion_mutliplier` parameter is present.  You can see the issue by printing out the step function parameters and seeing they do not change when the motion multiplier is added.   This is caused by this line: https://github.com/gtri/scrimmage/blob/master/src/simcontrol/SimControl.cpp#L1553 where the controller step function is fed `t` and `dt` instead of `temp_t` and `motion_dt`, even though it is run within the motion multiplier subloop.   The fix seems to be just changing those variables in the function call. 

* what scrimmage commit are you on? b696da4 

* describe any changes you made to scrimmage. N/A

* If you can recreate the issue using only plugins in the scrimmage repository, what mission file are you running? N/A

kemen209, 2022-07-13:
I had same doute on this issue. The usage on t_/dt_ start from commit add loop_rate api to plugins 6e65a69c287c05551a9f770267d7e5ceb4c9888a, but the message didn't given out any though on why changed it.

```
         // run controllers in a single thread since they are serially connected
         for (EntityPtr &ent : ents_) {
             for (auto c : ent->controllers()) {
-                success &= exec_step(c, [&](auto c){return c->step(temp_t, motion_dt);});
+                success &= exec_step(c, [&](auto c){
+                  return c->step_loop_timer(dt_) ?
+                    c->step(t_, dt_) : true;});
             }
         }
```

kemen209, 2022-07-13:
@shaun-d-anderson @esquires Hi, any idea on this?

kemen209, 2022-07-14:
> What interesting me is that the SimpleCar model will work smoothly with this change. If i change the t_/dt_ back to temp_t/motion_dt, the SimpleCar model will behave strangely:
> 
> (1) desired_speed = 10 m/s, desired_heading = 45 deg, start to turn immediately. (2) desired_speed = 12 m/s, desired_heading = 45 deg, it will start to turn after 200+ seconds!!!

After i done some digging in PID controller, this phenomenon will solved by using another set to PID value(lower D factor to 1/10 would work). So i will stick to use temp_t/motion_dt is the right way to go.

waford, 2025-05-06:
Addressed in #606

closed 2025-11-24 by wfsyre