gtri/scrimmage #539: Initial setup Issue (cmake .. / cmake generate step)

tsingh1201, 2022-01-19:
Answer the following questions:

* what are you trying to do?
First time user of SCRIMMAGE, currently running on Ubuntu 20.04.3 LTS
Followed all steps, got stuck on step "cmake .." and had to manually install the following (as suggested by terminal):
librapidxml-dev libeigen3-dev libgeographic-dev libboost-thread-dev libboost-date-time-dev libboost-graph-dev libboost-iostreams-dev libboost-program-options-dev libboost-regex-dev libboost-filesystem-dev libboost-system-dev autoconf automake libtool curl unzip ccache parallel libbullet-dev graphviz doxygen libopencv-dev libvtk6-dev tcl-vtk7 python python-setuptools python-numpy python-dev python-pip python-wxmpl python-matplotlib python-pandas python-wxtools python3 python3-setuptools python3-numpy python3-dev python3-pip python3-venv python3-matplotlib python3-pandas

All were able to be installed except for:
python-pip (which says it is replaced by python3-pip)
python-matplotlib (which I was able to install through git clone https://github.com/matplotlib/matplotlib.git)
python-wxmpl (was not able to find a way for the file to be found)

* what is the problem and how can it be recreated?
I was able to solve the initial problem of lacking a large amount of the files by manually installing, however I get this error now when running "cmake .." (error shown below) 

**-- The imported target "pvtk" references the file
   "/usr/bin/pvtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- Found VTK Version: 7.1.1
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GRPCPP_LIBRARY
    linked by target "scrimmage-protos" in directory /home/taaren/scrimmage/scrimmage/src/proto

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.**

* what scrimmage commit are you on? You can see this with `git rev-parse HEAD`
566ef92de67ac7df0b18e5482ee336ebfc6763ca

* describe any changes you made to scrimmage.
* None (just manually installed listed files)

* If you can recreate the issue using only plugins in the scrimmage repository, what mission file are you running?
Issue regarding setup, cannot figure out how to have GRPCPP_LIBRARY set to FOUND

SyllogismRXS, 2022-01-20:
On Ubuntu 20.04, you should only be installing python3 packages. For example: python3-matplotlib. You don't need "wxmpl". 

The CMake error you are getting is about your system not having GRPC. Did you install scrimmage-grpc and scrimmage-dependencies?

tsingh1201, 2022-01-20:
I installed the scrimmage dependencies using sudo apt-get install scrimmage-dependencies scrimmage-jsbsim and got the following:

scrimmage-dependencies is already the newest version (0.2.2.8-8-focalppa0).
The following packages were automatically installed and are no longer required:
  cuda-nvrtc-11-3 cuda-nvrtc-dev-11-3 cython3 docutils-common docutils-doc
  fltk1.3-doc fluid gazebo11 gazebo11-common gazebo11-plugin-base
  gir1.2-mutter-2 google-mock googletest ignition-tools lib32asan4
  lib32cilkrts5 lib32gcc1 lib32mpx2 lib32ubsan0 libapr1-dev libaprutil1-dev
  libargon2-0 libarmadillo8 libassimp-dev libassimp5 libassuan-dev
  libavdevice-dev libavfilter-dev libavresample3 libavutil55 libb64-0d
  libbind9-160 libbondcpp1d libboost-all-dev libboost-atomic-dev
  libboost-chrono-dev libboost-container-dev libboost-context-dev
  libboost-coroutine-dev libboost-date-time1.65.1 libboost-exception-dev
  libboost-fiber-dev libboost-filesystem1.65.1 libboost-graph-parallel-dev
  libboost-iostreams1.65.1 libboost-locale-dev libboost-log-dev
  libboost-math-dev libboost-mpi-dev libboost-mpi-python-dev
  libboost-numpy-dev libboost-python-dev libboost-random-dev
  libboost-serialization-dev libboost-stacktrace-dev libboost-system1.65.1
  libboost-test-dev libboost-thread1.65.1 libboost-timer-dev
  libboost-tools-dev libboost-type-erasure-dev libboost-wave-dev libbz2-dev
  libcapnp-0.6.1 libccd-dev libccd2 libcdio17 libcharls1 libclass-loader0d
  libclass-loader1d libconsole-bridge-dev libconsole-bridge0.4 libcpp-common0d
  libdart-collision-bullet-dev libdart-collision-ode-dev libdart-dev
  libdart-external-ikfast-dev libdart-external-odelcpsolver-dev
  libdart-utils-dev libdart-utils-urdf-dev libdart6 libdart6-collision-bullet
  libdart6-collision-ode libdart6-external-odelcpsolver libdart6-utils
  libdart6-utils-urdf libdns-export1100 libdns1100 libdouble-conversion1
  libdrm-dev libdynamic-reconfigure-config-init-mutex0d libevent-2.1-6
  libfcl-dev libfcl0.5 libflann-dev libflann1.9 libfltk-cairo1.3
  libfltk-forms1.3 libfltk-gl1.3 libfltk-images1.3 libfltk1.3 libfltk1.3-dev
  libfreeimage-dev libfreeimage3 libgazebo11 libgazebo11-dev libgdbm5
  libgdcm2.8 libgeos-3.6.2 libgeotiff2 libgles2-mesa libglew2.0
  libglvnd-core-dev libgmime-3.0-0 libgnome-desktop-3-17 libgpg-error-dev
  libgpgme-dev libgspell-1-1 libgtest-dev libgts-dev libhwloc5
  libignition-cmake2-dev libignition-common3 libignition-common3-av
  libignition-common3-av-dev libignition-common3-core-dev
  libignition-common3-dev libignition-common3-events
  libignition-common3-events-dev libignition-common3-graphics
  libignition-common3-graphics-dev libignition-common3-profiler
  libignition-common3-profiler-dev libignition-fuel-tools4
  libignition-fuel-tools4-dev libignition-math6 libignition-math6-dev
  libignition-msgs5 libignition-msgs5-dev libignition-tools-dev
  libignition-transport8 libignition-transport8-core-dev
  libignition-transport8-dev libignition-transport8-log
  libignition-transport8-log-dev libimage-transport0d libip4tc0 libip6tc0
  libiptc0 libisc-export169 libisc169 libisccc160 libisccfg160 libisl19
  libjs-sphinxdoc libjs-underscore libjson-c3 libjxr0 libllvm11 liblog4cxx-dev
  liblog4cxx10v5 liblouis14 liblouisutdml8 liblwres160 liblz4-dev
  libmessage-filters1d libmozjs-52-0 libmutter-2-0 libmysofa0 libmysqlclient20
  libnorm-dev libnss-myhostname libnvidia-common-495 liboctomap-dev
  liboctomap1.9 libode-dev libode8 libogre-1.9-dev libogre-1.9.0v5
  libopenblas-base libopencv-core3.2 libopencv-flann3.2 libopencv-imgproc3.2
  libopencv-ml3.2 libopencv-photo3.2 libopencv-shape3.2 libopencv-video3.2
  libopenni-dev libopenni-sensor-pointclouds0 libopenni0 libopenni2-0
  libopenni2-dev liborcus-0.13-0 liborocos-kdl-dev liborocos-kdl1.4
  libpcl-apps1.10 libpcl-common1.10 libpcl-features1.10 libpcl-filters1.10
  libpcl-io1.10 libpcl-kdtree1.10 libpcl-keypoints1.10 libpcl-ml1.10
  libpcl-octree1.10 libpcl-outofcore1.10 libpcl-people1.10
  libpcl-recognition1.10 libpcl-registration1.10 libpcl-sample-consensus1.10
  libpcl-search1.10 libpcl-segmentation1.10 libpcl-stereo1.10
  libpcl-surface1.10 libpcl-tracking1.10 libpcl-visualization1.10 libpgm-dev
  libpoco-dev libpococrypto62 libpocodata62 libpocodatamysql62
  libpocodataodbc62 libpocodatasqlite62 libpocoencodings62 libpocofoundation50
  libpocofoundation62 libpocojson62 libpocomongodb62 libpoconet62
  libpoconetssl62 libpocoredis62 libpocoutil62 libpocoxml62 libpocozip62
  libpoppler73 libpostproc-dev libpostproc54 libpotrace0 libprocps6 libproj12
  libprotobuf-lite10 libprotobuf10 libprotoc-dev libpyside2-dev
  libpyside2-py3-5.14 libpython-all-dev libpython3.6 libpython3.6-dev
  libpython3.6-minimal libpython3.6-stdlib libqpdf21 libqt5concurrent5
  libqt5designercomponents5 libqt5opengl5-dev libqt5quickparticles5
[... 110 lines cut ...]
  ros-noetic-rqt-rviz ros-noetic-rqt-service-caller ros-noetic-rqt-shell
  ros-noetic-rqt-srv ros-noetic-rqt-tf-tree ros-noetic-rqt-top
  ros-noetic-rqt-topic ros-noetic-rqt-web ros-noetic-rviz
  ros-noetic-rviz-plugin-tutorials ros-noetic-rviz-python-tutorial
  ros-noetic-self-test ros-noetic-sensor-msgs ros-noetic-shape-msgs
  ros-noetic-simulators ros-noetic-smach ros-noetic-smach-msgs
  ros-noetic-smach-ros ros-noetic-smclib ros-noetic-stage ros-noetic-stage-ros
  ros-noetic-std-msgs ros-noetic-std-srvs ros-noetic-stereo-image-proc
  ros-noetic-stereo-msgs ros-noetic-tf ros-noetic-tf-conversions
[... cut ...]
  ros-noetic-vision-opencv ros-noetic-visualization-marker-tutorials
  ros-noetic-visualization-msgs ros-noetic-visualization-tutorials
  ros-noetic-viz ros-noetic-webkit-dependency ros-noetic-xacro
  ros-noetic-xmlrpcpp sbcl sdformat9-sdf shiboken2 shim sip-dev tcl-dev
  tcl8.6-dev tk-dev tk8.6-dev ttf-dejavu-core vtk7 x11proto-composite-dev
  x11proto-damage-dev x11proto-fixes-dev x11proto-kb-dev
  x11proto-xf86vidmode-dev

I don't see scrimmage-grpc in the list, is there any way to install it? (sudo get-apt install didn't work)

tsingh1201, 2022-01-20:
Update: 
when trying sudo apt-get install scrimmage-grpc , I get the following:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package scrimmage-grpc

No clue why it is not working, but I checked https://launchpad.net/~kevin-demarco/+archive/ubuntu/scrimmage?field.series_filter=focal and see the package available. I tried the in-terminal ppa method and manually adding it to my system's source codes, still no luck. Is there any way to make it possible to access the scrimmage-grpc package?

SyllogismRXS, 2022-01-20:
I forgot that on 20.04, Ubuntu provides a version of grpc through their own package manager, so we didn't have to provide a custom grpc package through Launchpad. Try installing grpc on Ubuntu 20.04 with the command: `sudo apt-get install libgrpc-dev`

tsingh1201, 2022-01-20:
Just tried it, unfortunately I got the same result, is there possibly any way to go into the scripts and change what scrimmage is looking for as FOUND? Below is the error I keep getting:

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GRPCPP_LIBRARY
    linked by target "scrimmage-protos" in directory /home/taaren/scrimmage/scrimmage/src/proto

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

shaun-d-anderson, 2022-03-15:
The libgrpc++-dev package is what is needed fro focal install.  I have updated the install dependencies script.  mainly you should be able to `sudo apt-get install libgrpc-dev libgrpc++-dev`

closed 2022-03-15 by shaun-d-anderson