gtri/scrimmage #549: Looking for scrimmage/proto/Frame.pb.h and ../../Shape.pb.h

arschoon, 2022-06-09:
Answer the following questions:

* what are you trying to do?
I am looking for scrimmage/proto/Frame.pb.h and scrimmage/proto/Shape.pb.h . These two are dependencies for scrimmage/network/interface.h and scrimmage/viewer/Updater.h, respectively.
* what is the problem and how can it be recreated?
I am not able to find where I can download these two files as I am having a hard time finding a package that will install them for me. 

* what scrimmage commit are you on? You can see this with `git rev-parse HEAD`
* -This is a local repo copy

* describe any changes you made to scrimmage.
I am making a change to the output of viewer.cpp, no other files have been modified. I am only trying to test my changes but the #include errors form the missing .h files from the first question are stopping the build.

* If you can recreate the issue using only plugins in the scrimmage repository, what mission file are you running?

tm132, 2022-11-17:
The main proto files are located here:
`scrimmage/src/proto/scrimmage/proto`

During the build process, assuming you're building from a directory `scrimmage/build/`, the `Frame.pb.h` et al get generated from those proto files, and placed here:
- header files (e.g., `Frame.pb.h`): `scrimmage/build/src/proto/scrimmage/proto/`
- python files (e.g., `Frame_pb2.py`): `scrimmage/python/scrimmage/proto`/

Libraries that have dependencies on these should have dependencies listed in their CMakeLists.txt file:
```
add_dependencies(${LIBRARY_NAME}
  scrimmage-proto-gen [...]
)
```
And should link to them in their `CMakeLists.txt` configuration:
```
target_link_libraries(${LIBRARY_NAME}
  PUBLIC
    scrimmage-protos
   [...]
```

wfsyre, 2025-01-08:
These files are generated by the build, closing issue

closed 2025-01-08 by wfsyre