gtri/scrimmage #580: I'm having issues with $make when building SCRIMMAGE

Emelles, 2023-10-06:
Answer the following questions:

* what are you trying to do?

I am trying to build SCRIMMAGE on machine. Ubuntu 20.04.06 LTS 64-bit 

* what is the problem and how can it be recreated?

I am following the directions in the README.md. 
Everything seems to be running smoothly until the section 
```
$ mkdir build && cd build
$ cmake ..
$ make
```

I encounter an error with $ make. I've copied it here:

_[  4%] Built target scrimmage-proto-gen
[  4%] Building CXX object src/proto/CMakeFiles/scrimmage-protos.dir/scrimmage/proto/Color.pb.cc.o
In file included from /home/Emelles/scrimmage/build/src/proto/scrimmage/proto/Color.pb.cc:4:
/home/Emelles/scrimmage/build/src/proto/scrimmage/proto/Color.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/proto/CMakeFiles/scrimmage-protos.dir/build.make:231: src/proto/CMakeFiles/scrimmage-protos.dir/scrimmage/proto/Color.pb.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3063: src/proto/CMakeFiles/scrimmage-protos.dir/all] Error 2
make: *** [Makefile:152: all] Error 2_

I don't know what to make of this error message.
Checking the protobuf installation 
`$ sudo apt install protobuf-compiler`
_Reading package lists... Done
Building dependency tree       
Reading state information... Done
protobuf-compiler is already the newest version (3.6.1.3-2ubuntu5.2).
The following packages were automatically installed and are no longer required:
  gir1.2-goa-1.0 libfwupdplugin1 libllvm11 libxmlb1 scrimmage-pybind11
Use 'sudo apt autoremove' to remove them.
0 to upgrade, 0 to newly install, 0 to remove and 26 not to upgrade._

Reviewing the previous step, `$ source /opt/scrimmage/*/setup.sh`. It successfully creates a .scrimmage directory in my HOME. If something went wrong in the earlier steps, they're not screaming errors at me.

I'd appreciate any insight into this issue. 

* what scrimmage commit are you on? You can see this with `git rev-parse HEAD`

My output for this command is 'bde1ed5422a8c0380986587cdb87bd11c7a321c4'

* describe any changes you made to scrimmage.

None, still not installed.

wfsyre, 2023-10-11:
This is likely a 20.04 problem. I updated the build instructions for 20.04 today. Could you go through them and see if it changes anything for you?

Emelles, 2023-10-17:
> This is likely a 20.04 problem. I updated the build instructions for 20.04 today. Could you go through them and see if it changes anything for you?

Thanks for the update, but I still have the issue after running your new stuff.

My output for cmake .. shows:

*-- Python Versions Found: 3.9.12
-- PYTHON_LIBRARIES : /home/Emelles/anaconda3/lib/libpython3.9.so
-- PYTHON_INCLUDE_DIRS: /home/Emelles/anaconda3/include/python3.9
CMake Warning at /usr/share/cmake-3.16/Modules/FindProtobuf.cmake:499 (message):
  Protobuf compiler version 3.19.1 doesn't match library version 3.6.1
Call Stack (most recent call first):
  CMakeLists.txt:144 (find_package)

-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1") 
CMake Warning at /usr/share/cmake-3.16/Modules/FindProtobuf.cmake:499 (message):
  Protobuf compiler version 3.19.1 doesn't match library version 3.6.1
Call Stack (most recent call first):
  CMakeLists.txt:150 (find_package)

-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.6.1", minimum required is "3.6.1") 
CMake Warning (dev) at CMakeLists.txt:183 (find_package):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Environment variable JSBSIM_ROOT is set to:

    /opt/scrimmage/x86_64-linux-gnu/etc/JSBSim

  For compatibility, CMake is ignoring the variable.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found JSBSIM 
-- The imported target "vtkHashSource" references the file
   "/usr/bin/vtkHashSource-6.3"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-6.3"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkEncodeString" references the file
   "/usr/bin/vtkEncodeString-6.3"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- Found VTK Version: 6.3.0
-- Configuring done
CMake Warning at share/scrimmage/CMakeLists.txt:14 (add_executable):
  Cannot generate a safe runtime search path for target scrimmage-bin because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libpng16.so.16] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/Emelles/anaconda3/lib
    runtime library [libtiff.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/Emelles/anaconda3/lib
    runtime library [libfreetype.so.6] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/Emelles/anaconda3/lib
    runtime library [libz.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/Emelles/anaconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at /opt/scrimmage/x86_64-linux-gnu/share/cmake/pybind11/pybind11Tools.cmake:131 (add_library):
  Cannot generate a safe runtime search path for target py_bindings because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libpng16.so.16] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/Emelles/anaconda3/lib
    runtime library [libtiff.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/Emelles/anaconda3/lib
    runtime library [libfreetype.so.6] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/Emelles/anaconda3/lib
    runtime library [libz.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/Emelles/anaconda3/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  python/scrimmage/bindings_src/CMakeLists.txt:8 (pybind11_add_module)

-- Generating done
-- Build files have been written to: /home/Emelles/scrimmage/build *

This error still comes from make:

*[  4%] Built target scrimmage-proto-gen
[  4%] Building CXX object src/proto/CMakeFiles/scrimmage-protos.dir/scrimmage/proto/Color.pb.cc.o
In file included from /home/Emelles/scrimmage/build/src/proto/scrimmage/proto/Color.pb.cc:4:
/home/Emelles/scrimmage/build/src/proto/scrimmage/proto/Color.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
   10 | #include <google/protobuf/port_def.inc>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/proto/CMakeFiles/scrimmage-protos.dir/build.make:231: src/proto/CMakeFiles/scrimmage-protos.dir/scrimmage/proto/Color.pb.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3063: src/proto/CMakeFiles/scrimmage-protos.dir/all] Error 2
make: *** [Makefile:152: all] Error 2*

I'm really not certain where to begin with this error. Everything with protobuf appears to be fine?

Many thanks for your help.

Emelles, 2023-10-31:
I'm still struggling with this issue. I've re-installed from scratch and come across the exact same error. I'm gravitating to this issue in the terminal output -

_-- Found PythonInterp: /home/Emelles/anaconda3/bin/python (found version "3.9.12")_ 
_-- Found PythonLibs: /home/Emelles/anaconda3/lib/libpython3.9.so (found version "3.9.12")_ 
_-- Python Versions Found: 3.9.12_
_-- PYTHON_LIBRARIES : /home/Emelles/anaconda3/lib/libpython3.9.so_
_-- PYTHON_INCLUDE_DIRS: /home/Emelles/anaconda3/include/python3.9_
_CMake Warning at /usr/share/cmake-3.16/Modules/FindProtobuf.cmake:499 (message):_
  _Protobuf compiler version 3.19.1 doesn't match library version 3.6.1_
_Call Stack (most recent call first):_
  _CMakeLists.txt:147 (find_package)_

The fact that the library and compiler versions of Protobuf don't match appears to be the only issue that's been flagged that could lead to the subsequent error message after calling $make

_[  4%] Building CXX object src/proto/CMakeFiles/scrimmage-protos.dir/scrimmage/proto/Color.pb.cc.o_
_In file included from /home/Emelles/scrimmage/build/src/proto/scrimmage/proto/Color.pb.cc:4:_
_/home/Emelles/scrimmage/build/src/proto/scrimmage/proto/Color.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory_
   _10 | #include <google/protobuf/port_def.inc>_
      _|          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~_
_compilation terminated._
_make[2]: *** [src/proto/CMakeFiles/scrimmage-protos.dir/build.make:231: src/proto/CMakeFiles/scrimmage-protos.dir/scrimmage/proto/Color.pb.cc.o] Error 1_
_make[1]: *** [CMakeFiles/Makefile2:3063: src/proto/CMakeFiles/scrimmage-protos.dir/all] Error 2_
_make: *** [Makefile:152: all] Error 2_

It seems to me that this Protobuf error must be upstream of this version clash thing, as what would cause the two to not match to begin with?

Emelles, 2023-11-01:
Ok an update. 

I fixed that particular problem by changing the `$cmake ..` step to `$cmake .. -D Protobuf_PROTOC_EXECUTABLE=/usr/bin/protoc` as suggested in a stackoverflow thread. Then $make got past 4%. It got to 38% before stopping due to a new error in the terminal:

> [ 38%] Linking CXX executable ../../bin/scrimmage
> /usr/bin/ld: /usr/lib/x86_64-linux-gnu/libSM.so: undefined reference to `uuid_generate@UUID_1.0'
> /usr/bin/ld: /usr/lib/x86_64-linux-gnu/libSM.so: undefined reference to `uuid_unparse_lower@UUID_1.0'
> collect2: error: ld returned 1 exit status
> make[2]: *** [share/scrimmage/CMakeFiles/scrimmage-bin.dir/build.make:152: bin/scrimmage] Error 1
> make[1]: *** [CMakeFiles/Makefile2:2920: share/scrimmage/CMakeFiles/scrimmage-bin.dir/all] Error 2
> make: *** [Makefile:152: all] Error 2

Searching for libuuid `$apt-cache search libuuid` shows it's installed in my machine. 

So now the issue is with a non-human readable .so file. And it's only an undefined reference? I still can't help but feel that this issue and the previous issue were both due to something earlier in the process not working correctly, without notifying me. Something not being put in place is causing problems here. Although I might be going crazy.

I'd appreciate any input.

wfsyre, 2025-03-03:
We have a specific branch for 20.04 now. Can you try a clean install following the instructions and code on that branch. It looks like you have the wrong version of protobuf on you machine. What do you get when you run protoc --version?

waford, 2025-05-06:
Closing Issues as it appears stale.

closed 2025-05-06 by waford